ros-industrial/fanuc #145: support: add Gazebo simulation models

gavanderhoorn, 2015-01-20:
A nice-to-have: add Gazebo compatible versions of the manipulator xacros/urdfs in the support package so they can be placed in simulated worlds.

gavanderhoorn, 2016-02-10:
Might also be nice to somehow support [The Construct](http://www.theconstructsim.com/).

wkentaro, 2016-06-01:
Hi, I'm trying to do this because I need to simulate robot control for my project in Google Summer of Code in this year.
I worked around and have done 1. spawning model and 2. control joint position with ROS topic interface as shown below. [code](https://github.com/wkentaro/fanuc_experimental/tree/simulate-on-gazebo)
Maybe the joint controller should be changed to receive joint trajectory and controlled with actionlib, in normal cases (I have used PR2 and Baxter robots).
Do you have idea about this?  I'm new to ros-industrial so please correct me if I'm missing something.

[image]

gavanderhoorn, 2016-06-01:
Hi. Thanks for looking into this.

We've experimented with Gazebo packages for the models in ROS-Industrial some weeks/months ago and this is what we noticed:

 - position based control seems to be the most stable and usable:
   - it's closest to what most industrial robots support
   - doesn't seem influenced too much by guestimation of inertias and masses 
   - also doesn't require too much tuning of PIDs
   - it works relatively consistently across multiple versions of Gazebo
 - it pays off and is sort-of doable to keep support for a particular simulator (like Gazebo, but also V-REP fi) separate from the base robot support package, by making the base package contribute just the kinematics, geometry and the minimum dynamics bits, and have all the simulator specific parts be added by an add-on package (so `fanuc_lrmate200id_support` is the base package, `fanuc_lrmate200id_gazebo` packages the Gazebo support)

Structuring and naming the launch files is something that is rather subjective unfortunately, and I don't think we've had enough exposure to really standardise anything.

The two main approaches (which are very similar) can be found in the `_gazebo` packages in the [abb_experimental](https://github.com/ros-industrial/abb_experimental) and the [staubli_experimental](https://github.com/ros-industrial/staubli_experimental) repositories. I would appreciate it if you could take a look at those packages and see whether you have any thoughts on if things should be done differently.

Note that the proposed approach is rather ROS focused: still using URDF, only a single robot in isolation, etc. That might be something to address.

gavanderhoorn, 2016-06-01:
Also: because of the fact that inertias and masses are really only (gu)estimates, the dynamic behaviour will never really come close to the real robot. I feel this is important to mention, and so at least the Staubli packages include a warning about that ([here](https://github.com/ros-industrial/staubli_experimental/blob/18ac65912a704f8bfb5ec576b7f094956d226c87/staubli_tx90_support/package.xml#L33-L37) for the TX90 fi).

wkentaro, 2016-06-06:
> Structuring and naming the launch files is something that is rather subjective unfortunately, and I don't think we've had enough exposure to really standardise anything.
The two main approaches (which are very similar) can be found in the _gazebo packages in the abb_experimental and the staubli_experimental repositories. I would appreciate it if you could take a look at those packages and see whether you have any thoughts on if things should be done differently.

Thanks for the suggestion, I decided to follow the `_gazebo` name suffix to simply put gazebo stuffs into that package like the moveit related packages already in fanuc and fanuc_experimental repository.

jediofgever, 2019-10-14:
Currently, I am interested in this; 
progress can be seen in this branch ;
https://github.com/jediofgever/fanuc_experimental/tree/gazebo_dev

[image]

gavanderhoorn, 2020-08-30:
I'm going to close this.

Reason: creating these packages is quite some work, especially when we'd want to use `effort` interfaces everywhere -- and we should, as otherwise robot<->world/object interaction in Gazebo becomes very unstable/unpredictable. Effort interfaces would need tuning of PIDs and other such series/variant specific parameters, and with the many models we already support in the repository, that would mean *a lot* of tuning.

Creating Gazebo support packages for all robots therefore is not something we'll do 'just because we can', but will have to be prioritised based on when it is requested.

closed 2020-08-30 by gavanderhoorn