ros-industrial/fanuc #148: wiki: move tutorials to github repository

gavanderhoorn, 2015-02-03:
Similar to how catkin & MoveIt are documented.

It would probably make sense to wait for / contribute some way for in-vcs tutorials to be nicely integrated into the ROS wiki before doing this.

130s, 2015-03-21:
+1. 

We're on the same boat to seek how to do this. [This ticket](https://github.com/tork-a/rtmros_nextage/issues/12#issuecomment-72354678) might be of your interest (but there may not be really a new info).

gavanderhoorn, 2015-03-23:
Thanks for the comment. I've discussed this a bit with @shaun-edwards. Personally, I think a short term solution could be to create a new ROS wiki macro that can pull in externally generated MoinMoin. for MoinMoin generation from markdown (fi), something like pandoc could be used (although it currently doesn't seem to support MoinMoin directly. [this](http://marksy.arc90.com/learn) does though).

More long-term would be to go to an entirely different wiki engine. [gitit](http://gitit.johnmacfarlane.net/) seems like a nice example of an engine with support for more markup formats than just its own dialect.

@wjwwood wrote an interesting ROS2.0 whitepaper about this very subject recently btw. See [ROS 2 Documentation System](http://design.ros2.org/articles/ros_documentation_system.html).

gavanderhoorn, 2016-05-27:
Did some testing with Sphinx [here](http://gavanderhoorn.github.io/fanuc-doc-test). Work in progress, but I like it so far.

Doesn't integrate with the ROS wiki, but I feel the advantages outweigh that disadvantage.

130s, 2016-06-06:
Isn't this worth discussing on broader community, like http://discourse.ros.org/?

gavanderhoorn, 2016-06-06:
We can do that, but I first wanted to test out some things for myself, hence the test repository.