ros-industrial/fanuc #156: Improve MoveIt planning performance in Indigo

shaun-edwards, 2015-03-24:
Set planning attempts in MoveIt to "1" in order to improve planner performance.  See ros-industrial/universal_robot#193 for more info/rationale.

gavanderhoorn, 2015-05-05:
I can do this before an Indigo release, but I'm not sure whether this is actually needed for the fanuc robots we support.

I don't really know whether it has any influence on the OMPL/MoveIt code paths chosen, but the URx robots don't have a spherical wrist, which could be a complicating factor. The LWR+ does though, afaict.

Also, I haven't received any complaints about planning time from users.

gavanderhoorn, 2015-05-24:
I'm not sure this is an issue just on Indigo: without setting *Planning Attempts* to `1` I get some really erratic behaviour under Hydro as well.

gavanderhoorn, 2015-05-27:
Hm. `RRTConnect` always 'works' for me (ie: no strange paths, instant planning), irrespective of what I set *Planning Attempts* to.

Removing the `projection_evaluator` line from the yaml file makes any planner that depends on that fail, but only when you actually start using it (so not at system init). It also means that the default MoveIt config we ship is essentially broken for those planners. I don't like that too much.

Setting *Planning Attempts* to 1 seems to be the least invasive of the options @marcoesposito1988 describes in ros-industrial/universal_robot#193, but appears to not solve anything for robots that don't work with `LBKPIECE1` as a planner.

Ideally we'd set the default planner to `RRTConnect`, but seems to be impossible.

gavanderhoorn, 2015-09-11:
Closed in #175.

closed 2015-09-11 by gavanderhoorn

gavanderhoorn, 2015-11-25:
Configuring a default planner in a MoveIt configuration is now actually possible with the merge of ros-planning/moveit_ros#625.

gavanderhoorn, 2016-06-16:
I changed the default planner for all MoveIt configurations to be `RRTConnect`.