ros-industrial/fanuc #159: Add support for CR-35iA

gavanderhoorn, 2015-05-05:
As per title.

As a collaborative robot, it would seem to be an ideal robot for ROS to support.

walterpintor, 2016-06-01:
Hi @gavanderhoorn. May I know if there is any progress to complete this cobot?

gavanderhoorn, 2016-06-01:
Sure.

I have the support package done locally, just haven't released it yet. I just need to polish some things (mesh size for one: because of all the rounded geometry, even convex hulls are quite large).

VictorLamoine, 2016-06-01:
Gijs, tell me if you want me to help simplify the meshes and/or test locally.

walterpintor, 2016-06-01:
Awesome. If there is anything I can help with, let me know. 
We'd like to display cobot's capabilities using ROS Industrial here.

gavanderhoorn, 2016-06-03:
@walterpintor: I've just pushed the basic support package for the CR-35iA to `fanuc_experimental`. It should be good to go, but please pay attention to the note in the package manifest.

Let me know if you run into any issues using it.

@VictorLamoine: thanks for the offer, but I think you have more urgent things to do :). Much appreciated though :+1:.

closed 2016-06-03 by gavanderhoorn

walterpintor, 2016-06-04:
Thank you guys. I will give it a try and if there is any problem I will let you know.

walterpintor, 2016-06-04:
Just ran it using hydro and runs nice. will change it to green colors. You all have a great weekend

gavanderhoorn, 2016-06-04:
@walterpintor: if at all possible, please use something more recent than Hydro. ROS-Industrial (and these packages) have come a long way since then ..

gavanderhoorn, 2016-06-04:
@walterpintor: I've fixed the colour of the model in ros-industrial/fanuc_experimental@df671741.

walterpintor, 2016-06-05:
Thanks bud. I also changed them locally here. Much appreciated. 

Not to sound too demanding but is there any plan in the future to improve the .stls to look more like the real robot?

gavanderhoorn, 2016-06-05:
> Not to sound too demanding but is there any plan in the future to improve the .stls to look more like the real robot? 

See [wiki/fanuc_driver/Troubleshooting - ROS - RViz only shows 'collision quality' models](http://wiki.ros.org/action/show/fanuc_driver/Troubleshooting?action=show&redirect=fanuc%2FTroubleshooting#RViz_only_shows_.27collision_quality.27_models).

walterpintor, 2016-06-06:
Got it. Let's hope Fanuc one day can provide permission to let us use it in ROS... Anyway, thanks for your support, we truly appreciate it.