ros-industrial/fanuc #171: Help with fanuc M710c

sudarshan85, 2015-07-30:
Hey,

I am working on getting the fanuc M710ic working with ROS. It uses a R30iB controller. I followed all the instructions and am running rosstate on the TP while I issue my ROS commands. I am able to use the files for 200ic and am able to read the robot states. I confirmed this by viewing the manipulator move in RVIZ when I jogged the robot. However, I am not able to send control commands to the robot, When I try plan and execute in moveit I always get "controller is taking too long, aborting move" error. IF I set the execution monitoring parameter to false, then nothing happens, moveit just keeps on "executing" it. This seems to indicate that either the command is not reaching the controller or the controller is not able to read the command correctly.

Is there any way to fix this problem? Does this problem have to do with me using the 200ic support files for a 700ic model? 

Also, I was wondering if there was a way to bypass the moveit trajectory planner and directly use the in-built IK solver on the fanuc robots, essentially by just issuing a location command through a string of something which the controller can read and use its own IK solver to do its movements.

Thanks.

gavanderhoorn, 2015-07-30:
> I am working on getting the fanuc M710ic working with ROS. It uses a R30iB controller. I followed all the instructions and am running rosstate on the TP while I issue my ROS commands. I am able to [...]

Ok, so steps 1 & 2 of [Running the ROS-Industrial programs on your Fanuc robot](http://wiki.ros.org/fanuc/Tutorials/Running) are successful? The robot visualised in RViz shows the same motion as the real one.

> However, I am not able to send control commands to the robot, 

Is this still with only `rosstate` running on the robot?

> When I try plan and execute in moveit I always get "controller is taking too long, aborting move" error. IF I set the execution monitoring parameter to false, then nothing happens, moveit just keeps [...]

Again: you're only running `rosstate` at this point?

> Is there any way to fix this problem?

If you answered *yes* to all my questions, then you should essentially move to step 3 of the tutorial I linked ("*On the TP, start the `ros` TPE program*"). In order for the robot to execute your trajectories, you need to be running the `ros` TP program. `rosstate`, as the name implies`, only broadcasts joint states, it cannot perform any motion.

> Does this problem have to do with me using the 200ic support files for a 700ic model? 

So no, it does not, as far as I can tell.

You will however run into issues with collision avoidance and path planning in Cartesian space. Without a proper model of your actual robot, MoveIt (or any planner) cannot plan correctly for your robot. It's entirely up to you, but I recommend to first create a support pkg for your particular robot.

> Also, I was wondering if there was a way to bypass the moveit trajectory planner and directly use the in-built IK solver on the fanuc robots, essentially by just issuing a location command through a string of something which the controller can read and use its own IK solver to do its movements.

No, not at this time. The `simple_message` protocol does not have support for Cartesian motions at this time, and that is what the `fanuc_driver` package uses to communicate with the programs running on the controller.

Proper Cartesian support is actually rather tricky, especially if you want to abstract over a heterogeneous set of motion controllers. A proposal to add this to `simple_message` was recently submitted though, see ros-industrial/rep#12.

An alternative might be to generate a set of ROS `Pose`s, use IK to turn those into joint space targets and send that as a `JointTrajectory` msg to the ROS-Industrial `joint_trajectory_action` node on `/joint_path_command`. You'll have to make sure your trajectory is smooth, all points are reachable by the robot, add timing info (ie: `time_from_start`) and do all collision checking yourself.

Another alternative might be to make use of something like [descartes](http://wiki.ros.org/descartes).

---

PS: could you please post general support questions like this either to the ROS-Industrial [mailing list](https://groups.google.com/forum/?fromgroups#!forum/swri-ros-pkg-dev) or on [ROS Answers](http://answers.ros.org)? Just tag it appropriately (`fanuc`, `fanuc_driver`, `ros_industrial`, etc) and I'll make sure to respond there. This issue tracker is mostly used for reporting actual defects and feature requests. Thank you.

gavanderhoorn, 2015-07-30:
> It's entirely up to you, but I recommend to first create a support pkg for your particular robot model and variant.

If you can tell me which particular variant of the M-710iC you have, I might have an unreleased support pkg for it.

sudarshan85, 2015-07-30:
Hi,

Thank you for your very detailed and quick reply. As per your request, I've moved this conversation to answers.ros.org here: http://answers.ros.org/question/214975/help-with-fanuc-m710ic-50/

I have answered your questions and have updated my situations. I will close this issue here so that we can continue our conversation there.

Thanks.

closed 2015-07-30 by sudarshan85