ros-industrial/fanuc #180: Release into Jade

davetcoleman, 2015-12-03:
MoveIt! is now using the fanuc m10ia for its example benchmark in its [benchmark suite](https://github.com/ros-planning/moveit_ros/pull/616), but the benchmark suite only runs on Jade. It would be great if this was available in jade as well. Thanks!

gavanderhoorn, 2015-12-04:
I'd be willing to do this, but I can't before `industrial_core` is released into Jade.

Afaik a Jade release of those packages is planned for January next year.

gavanderhoorn, 2015-12-04:
I'll mark this as blocked for now, and will come back to it once `industrial_core` is released.

shaun-edwards, 2015-12-04:
January is the date, but I can see about moving this date up since there is
a good reason.

davetcoleman, 2015-12-04:
Not a huge deal, I was just putting in my vote of support. Thanks for providing a date

gavanderhoorn, 2015-12-04:
Yeah, I figured as much.

Thanks for putting it on my TODO list.

gavanderhoorn, 2015-12-04:
First step: get docs build (ros/rosdistro#9960).

gavanderhoorn, 2016-04-01:
Can be done now that `industrial_core` is in Jade.

gavanderhoorn, 2017-10-04:
As Jade has been EOL-ed, we'll not release `fanuc` for it. Instead we'll focus on Kinetic.

closed 2017-10-04 by gavanderhoorn

davetcoleman, 2017-10-09:
correct. and jade is already EOL now.