ros-industrial/fanuc #185: CMake warnings with catkin_tools on Travis CI

VictorLamoine, 2016-04-18:
```bash
[fanuc_lrmate200ib_support] -- Configuring done 
[fanuc_lrmate200ib_support] CMake Warning (dev) at /opt/ros/indigo/share/catkin/cmake/test/tests.cmake:148 (add_dependencies): 
[fanuc_lrmate200ib_support]   Policy CMP0046 is not set: Error on non-existent dependency in 
[fanuc_lrmate200ib_support]   add_dependencies.  Run "cmake --help-policy CMP0046" for policy details. 
[fanuc_lrmate200ib_support]   Use the cmake_policy command to set the policy and suppress this warning. 
[fanuc_lrmate200ib_support]  
[fanuc_lrmate200ib_support]   The dependency target "fanuc_driver_motion_streaming_interface" of target 
[fanuc_lrmate200ib_support]   "_run_tests_fanuc_lrmate200ib_support_roslaunch-check_tests_roslaunch_test.xml" 
[fanuc_lrmate200ib_support]   does not exist. 
[fanuc_lrmate200ib_support] Call Stack (most recent call first): 
[fanuc_lrmate200ib_support]   /opt/ros/indigo/share/roslaunch/cmake/roslaunch-extras.cmake:35 (catkin_run_tests_target) 
[fanuc_lrmate200ib_support]   CMakeLists.txt:11 (roslaunch_add_file_check) 
[fanuc_lrmate200ib_support] This warning is for project developers.  Use -Wno-dev to suppress it. 
[fanuc_lrmate200ib_support] CMake Warning (dev) at /opt/ros/indigo/share/catkin/cmake/test/tests.cmake:148 (add_dependencies): 
[fanuc_lrmate200ib_support]   Policy CMP0046 is not set: Error on non-existent dependency in 
[fanuc_lrmate200ib_support]   add_dependencies.  Run "cmake --help-policy CMP0046" for policy details. 
[fanuc_lrmate200ib_support]   Use the cmake_policy command to set the policy and suppress this warning. 
[fanuc_lrmate200ib_support]   The dependency target "fanuc_driver_motion_streaming_interface_bswap" of 
[fanuc_lrmate200ib_support]   target 
[fanuc_lrmate200ib_support]   "_run_tests_fanuc_lrmate200ib_support_roslaunch-check_tests_roslaunch_test.xml" 
[fanuc_lrmate200ib_support]   does not exist. 
[fanuc_lrmate200ib_support] Call Stack (most recent call first): 
[fanuc_lrmate200ib_support]   /opt/ros/indigo/share/roslaunch/cmake/roslaunch-extras.cmake:35 (catkin_run_tests_target) 
[fanuc_lrmate200ib_support]   CMakeLists.txt:11 (roslaunch_add_file_check) 
[fanuc_lrmate200ib_support] This warning is for project developers.  Use -Wno-dev to suppress it. 
[fanuc_lrmate200ib_support] CMake Warning (dev) at /opt/ros/indigo/share/catkin/cmake/test/tests.cmake:148 (add_dependencies): 
[fanuc_lrmate200ib_support]   Policy CMP0046 is not set: Error on non-existent dependency in 
[fanuc_lrmate200ib_support]   add_dependencies.  Run "cmake --help-policy CMP0046" for policy details. 
[fanuc_lrmate200ib_support]   Use the cmake_policy command to set the policy and suppress this warning. 
[fanuc_lrmate200ib_support]   The dependency target "fanuc_driver_robot_state" of target 
[fanuc_lrmate200ib_support]   "_run_tests_fanuc_lrmate200ib_support_roslaunch-check_tests_roslaunch_test.xml" 
[fanuc_lrmate200ib_support]   does not exist. 
[fanuc_lrmate200ib_support] Call Stack (most recent call first): 
[fanuc_lrmate200ib_support]   /opt/ros/indigo/share/roslaunch/cmake/roslaunch-extras.cmake:35 (catkin_run_tests_target) 
[fanuc_lrmate200ib_support]   CMakeLists.txt:11 (roslaunch_add_file_check) 
[fanuc_lrmate200ib_support] This warning is for project developers.  Use -Wno-dev to suppress it. 
[fanuc_lrmate200ib_support] CMake Warning (dev) at /opt/ros/indigo/share/catkin/cmake/test/tests.cmake:148 (add_dependencies): 
[fanuc_lrmate200ib_support]   Policy CMP0046 is not set: Error on non-existent dependency in 
[fanuc_lrmate200ib_support]   add_dependencies.  Run "cmake --help-policy CMP0046" for policy details. 
[fanuc_lrmate200ib_support]   Use the cmake_policy command to set the policy and suppress this warning. 
[fanuc_lrmate200ib_support]   The dependency target "fanuc_driver_robot_state_bswap" of target 
[fanuc_lrmate200ib_support]   "_run_tests_fanuc_lrmate200ib_support_roslaunch-check_tests_roslaunch_test.xml" 
[fanuc_lrmate200ib_support]   does not exist. 
[fanuc_lrmate200ib_support] Call Stack (most recent call first): 
[fanuc_lrmate200ib_support]   /opt/ros/indigo/share/roslaunch/cmake/roslaunch-extras.cmake:35 (catkin_run_tests_target) 
[fanuc_lrmate200ib_support]   CMakeLists.txt:11 (roslaunch_add_file_check) 
[fanuc_lrmate200ib_support] This warning is for project developers.  Use -Wno-dev to suppress it. 
[fanuc_lrmate200ib_support] CMake Warning (dev) at /opt/ros/indigo/share/catkin/cmake/test/tests.cmake:148 (add_dependencies): 
[fanuc_lrmate200ib_support]   Policy CMP0046 is not set: Error on non-existent dependency in 
[fanuc_lrmate200ib_support]   add_dependencies.  Run "cmake --help-policy CMP0046" for policy details. 
[fanuc_lrmate200ib_support]   Use the cmake_policy command to set the policy and suppress this warning. 
[fanuc_lrmate200ib_support]   The dependency target "joint_trajectory_action" of target 
[fanuc_lrmate200ib_support]   "_run_tests_fanuc_lrmate200ib_support_roslaunch-check_tests_roslaunch_test.xml" 
[fanuc_lrmate200ib_support]   does not exist. 
[fanuc_lrmate200ib_support] Call Stack (most recent call first): 
[fanuc_lrmate200ib_support]   /opt/ros/indigo/share/roslaunch/cmake/roslaunch-extras.cmake:35 (catkin_run_tests_target) 
[fanuc_lrmate200ib_support]   CMakeLists.txt:11 (roslaunch_add_file_check) 
[fanuc_lrmate200ib_support] This warning is for project developers.  Use -Wno-dev to suppress it. 
[fanuc_lrmate200ib_support] CMake Warning (dev) at /opt/ros/indigo/share/catkin/cmake/test/tests.cmake:142 (add_dependencies): 
[fanuc_lrmate200ib_support]   Policy CMP0046 is not set: Error on non-existent dependency in 
[fanuc_lrmate200ib_support]   add_dependencies.  Run "cmake --help-policy CMP0046" for policy details. 
[fanuc_lrmate200ib_support]   Use the cmake_policy command to set the policy and suppress this warning. 
[fanuc_lrmate200ib_support]   The dependency target "fanuc_driver_motion_streaming_interface" of target 
[fanuc_lrmate200ib_support]   "run_tests_fanuc_lrmate200ib_support_roslaunch-check_tests_roslaunch_test.xml" 
[fanuc_lrmate200ib_support]   does not exist. 
[fanuc_lrmate200ib_support] Call Stack (most recent call first): 
[fanuc_lrmate200ib_support]   /opt/ros/indigo/share/roslaunch/cmake/roslaunch-extras.cmake:35 (catkin_run_tests_target) 
[fanuc_lrmate200ib_support]   CMakeLists.txt:11 (roslaunch_add_file_check) 
[fanuc_lrmate200ib_support] This warning is for project developers.  Use -Wno-dev to suppress it. 
[... 34 lines cut ...]
[fanuc_lrmate200ib_support]   "run_tests_fanuc_lrmate200ib_support_roslaunch-check_tests_roslaunch_test.xml" 
[fanuc_lrmate200ib_support]   does not exist. 
[fanuc_lrmate200ib_support] Call Stack (most recent call first): 
[fanuc_lrmate200ib_support]   /opt/ros/indigo/share/roslaunch/cmake/roslaunch-extras.cmake:35 (catkin_run_tests_target) 
[fanuc_lrmate200ib_support]   CMakeLists.txt:11 (roslaunch_add_file_check) 
[fanuc_lrmate200ib_support] This warning is for project developers.  Use -Wno-dev to suppress it. 
[fanuc_lrmate200ib_support]  
[fanuc_lrmate200ib_support] CMake Warning (dev) at /opt/ros/indigo/share/catkin/cmake/test/tests.cmake:142 (add_dependencies): 
[fanuc_lrmate200ib_support]   Policy CMP0046 is not set: Error on non-existent dependency in 
[fanuc_lrmate200ib_support]   add_dependencies.  Run "cmake --help-policy CMP0046" for policy details. 
[fanuc_lrmate200ib_support]   Use the cmake_policy command to set the policy and suppress this warning. 
[fanuc_lrmate200ib_support]  
[fanuc_lrmate200ib_support]   The dependency target "joint_trajectory_action" of target 
[fanuc_lrmate200ib_support]   "run_tests_fanuc_lrmate200ib_support_roslaunch-check_tests_roslaunch_test.xml" 
[fanuc_lrmate200ib_support]   does not exist. 
[fanuc_lrmate200ib_support] Call Stack (most recent call first): 
[fanuc_lrmate200ib_support]   /opt/ros/indigo/share/roslaunch/cmake/roslaunch-extras.cmake:35 (catkin_run_tests_target) 
[fanuc_lrmate200ib_support]   CMakeLists.txt:11 (roslaunch_add_file_check) 
[fanuc_lrmate200ib_support] This warning is for project developers.  Use -Wno-dev to suppress it. 
[fanuc_lrmate200ib_support]  
[fanuc_lrmate200ib_support] -- Generating done 
```

All of these warnings clutter CMake output in Travis builds; like here:
https://travis-ci.org/InstitutMaupertuis/ensenso_extrinsic_calibration/jobs/123913833#L8150

gavanderhoorn, 2016-04-18:
I'm not sure it will do much, but all of those `fanuc_*` targets are actually defined in `fanuc_driver`, which afaict is not part of your travis workspace (you only seem to list `fanuc_experimental` in your `.travis.yml` ([here](https://github.com/VictorLamoine/ensenso_extrinsic_calibration/blob/travis-ci/.travis.yml#L22))).

VictorLamoine, 2016-04-18:
Here is a big log:
https://gist.github.com/VictorLamoine/718b6c8fe0d8c48107ce183c55a26201

In fact the problem is in ROS CMake files;
```
/opt/ros/indigo/share/roslaunch/cmake/roslaunch-extras.cmake
/opt/ros/indigo/share/catkin/cmake/test/tests.cmake
```

dirk-thomas, 2016-04-18:
The problem here is that these lines (https://github.com/ros-industrial/fanuc/blob/705ffbed4fb19f109b4127e91c86843dd4673c7a/fanuc_driver/CMakeLists.txt#L77-L83) declare dependencies which are not targets if you don't build the code within the same workspace as its dependencies or with isolation. Instead the code should only declare the dependencies if these names are actual valid targets.

gavanderhoorn, 2016-04-18:
So am I to understand that the `DEPENDENCIES` argument is basically no longer usable then when building with `catkin_tools`? Or at least not like it's being used right now?

Any alternative approach to make sure ordering of targets is such that tests will not try to use targets that haven't been built yet?

dirk-thomas, 2016-04-18:
In CMake you shouldn't refer to targets as dependencies if they don't actually exist. The newer CMake versions are only more picky and telling you about it.

To support both cases (non-isolated as well as isolated builds) cleanly with a newer version of CMake you need to add the dependencies conditionally so only when the targets actually exist, e.g.:

```
set(deps "")
# not nice, but industrial_robot_client doesn't prefix its targets
if(TARGET joint_trajectory_action)
  list(APPEND deps joint_trajectory_action)
endif()
if(TARGET ${PROJECT_NAME}_robot_state)
  list(APPEND deps ${PROJECT_NAME}_robot_state)
endif()
...
roslaunch_add_file_check(tests/roslaunch_test.xml
    DEPENDENCIES ${deps})
```

gavanderhoorn, 2016-04-18:
But wouldn't that only work in the `CMakeLists.txt` of the `fanuc_driver` package itself?

My question was more about the `fanuc_driver_..` dependencies in the various `fanuc_x_support` packages: afaiu without exporting those targets they wouldn't exist when doing isolated builds even if they all share the same source space?

gavanderhoorn, 2016-04-18:
Btw: ROS Answers question this was based on [roslaunch_add_file_check(..) and build dependencies](http://answers.ros.org/question/125676).

dirk-thomas, 2016-04-18:
I am not sure if I understand the question correctly but `catkin_EXPORTED_TARGETS` always contains valid target names (isolated or not). If you build in isolation catkin will just create dummy targets for this to make your life easier: https://github.com/ros/catkin/blob/kinetic-devel/cmake/catkinConfig.cmake.in#L112-L118

commit 809602cf (2016-05-05): Remove roslaunch test DEPENDENCIES. Fix #185.

These targets don't exist when using isolated builds or when packages are
build that depend on binary distributions that provide the specified targets.

Newer versions of CMake will print copious warnings about the targets not
existing, so remove them.

Only the 'fanuc_driver' package is updated to detect whether the targets
exist in the workspace, to make sure its tests only run after its dependencies
have been built.
closed 2016-05-05 by gavanderhoorn