ros-industrial/fanuc #186: Add FAQ entry about J2-J3 coupling (and the built-in compensation)

gavanderhoorn, 2016-05-02:
It's not really obvious why the joint angles reported for J3 on the TP do not correspond to those in the `joint_states` msgs published by `robot_state`. Add a FAQ entry about this to the [Troubleshooting](http://wiki.ros.org/action/show/fanuc_driver/Troubleshooting) page.

gavanderhoorn, 2016-05-02:
Triggered by ros-industrial/fanuc_experimental#17 and ROS Answers [Strange behaviour from joint_trajectory_action to ABB robot](http://answers.ros.org/question/226402/strange-behaviour-from-joint_trajectory_action-to-abb-robot/).

simonschmeisser, 2016-09-05:
+1 from me, please also clarify the section "Outline" of http://wiki.ros.org/fanuc/Tutorials/Running, there is a slight hint there as to something being strange: "Verify J2-J3 coupling by jogging the robot in Roboguide. RVIZ should show same movements. " But it would be great to add another sentence like "The values of J2 and J3 are coupled in the controller (as well as on the TP) by J2' = J2-J3. fanuc_driver compensates this and joint values reported in ROS are thus independent." (I'm still sleepy so it might be the other way around ...)

(I requested wiki write permission but that has not happened so far)

gavanderhoorn, 2016-09-05:
(this is a general comment on this issue, and not directed towards @simonschmeisser)

This is always a bit of a dubious thing for me: on the one hand, adding this kind of information could save users some time in trying to understand what is going on.

On the other hand, I don't want the wiki pages to be so full of detail that people can use them as a step-by-step guide into the world of Fanuc robots (it applies to other brands as well tbh). These machines are far too dangerous for that, and I think users should get proper training *before* using anything like ROS with them (also, because if something 'bad' happens, this will reflect on ROS-Industrial, while it could well be that it is actually user error).

Any kind of training would have covered the J2-J3 coupling, and so that sentence on itself should already be enough to know what it is referring to, without adding more explanation to it.

Now I will obviously resolve this issue and add the requested information, but I do wanted to record somewhere that I'm starting to feel more and more uneasy about this 'trend'. In private emails I've had a significant nr of support requests from users that clearly never had used Fanuc robots before, nor had they received any kind of training, and were using the Installation and Configuration tutorials (which are far too detailed as well to my liking) as step-by-step guides, following the *magic incantations* to the letter.

gavanderhoorn, 2017-10-04:
Added entry about J2/J3 coupling to the FAQ in [fanuc_driver/Troubleshooting?action=diff&rev2=30&rev1=29](http://wiki.ros.org/action/info/fanuc_driver/Troubleshooting?action=diff&rev2=30&rev1=29).

gavanderhoorn, 2017-10-04:
Added a link to the new FAQ entry in the "run" tutorial in [fanuc/Tutorials/Running?action=diff&rev2=20&rev1=19](http://wiki.ros.org/action/info/fanuc/Tutorials/Running?action=diff&rev2=20&rev1=19) as per @simonschmeisser suggestion.

gavanderhoorn, 2017-10-04:
Thanks @simonschmeisser 👍

closed 2017-10-04 by gavanderhoorn