ros-industrial/fanuc #199: Kinetic devel

VictorLamoine, 2016-10-25:
Now that MoveIt has been released into kinetic-devel it would be nice to have a specific branch including fixes because the current IK Fast MoveIt plugins does not seem compatible with the MoveIt release. 
```bash
Errors     << fanuc_m10ia_moveit_plugins:make /home/im/code/catkin_workspace/logs/fanuc_m10ia_moveit_plugins/build.make.002.log                
In file included from /home/im/code/catkin_workspace/src/fanuc/fanuc_m10ia_moveit_plugins/m10ia_kinematics/src/fanuc_m10ia_manipulator_ikfast_moveit_plugin.cpp:49:0:
/opt/ros/kinetic/include/moveit/kinematics_base/kinematics_base.h:136:36: error: 'shared_ptr' in namespace 'std' does not name a template type
 MOVEIT_CLASS_FORWARD(KinematicsBase);
                                    ^
/opt/ros/kinetic/include/moveit/kinematics_base/kinematics_base.h:136:95: error: 'shared_ptr' in namespace 'std' does not name a template type
make[2]: *** [m10ia_kinematics/CMakeFiles/fanuc_m10ia_manipulator_moveit_ikfast_plugin.dir/src/fanuc_m10ia_manipulator_ikfast_moveit_plugin.cpp.o] Error 1
make[1]: *** [m10ia_kinematics/CMakeFiles/fanuc_m10ia_manipulator_moveit_ikfast_plugin.dir/all] Error 2
make: *** [all] Error 2
cd /home/im/code/catkin_workspace/build/fanuc_m10ia_moveit_plugins; catkin build --get-env fanuc_m10ia_moveit_plugins | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
```

Looks like C++11 not enabled.
Tested on Ubuntu 16.04 with ROS Kinetic shadow repositories

gavanderhoorn, 2016-10-25:
Yes, the IKFast plugins suffer from some bitrot. And they need to be regenerated anyway (#123).

I'll see if I can do something about that.

Pragmatic: in the meantime you could probably `CATKIN_IGNORE` them and use `trac_ik`.

gavanderhoorn, 2016-12-07:
@VictorLamoine: I've updated the title of this issue to better reflect the problem.

jettan, 2017-01-24:
Is this issue still present currently? I have successfully built all the moveit_plugins/ikfast plugins in this repo on kinetic.

gavanderhoorn, 2017-01-24:
I haven't changed anything, so if things now compile that is probably a result of something upstream accomodating me/us.

VictorLamoine, 2017-02-16:
We have problems in our CI because of these problems.

Maybe we could create a `kinetic-devel` branch and force enable C++11 with the `add_compile_options`?

anderwm, 2017-02-21:
Just had to add C++11 support (CMAKE_CXX_STANDARD 11) to the CMakeLists file to get the package generated by create_ikfast_moveit_plugin.py to build on Kinetic

commit c8eb156d (2017-05-31): moveit_plugins: C++11 for MoveIt on Kinetic. Fix #199.
closed 2017-06-02 by gavanderhoorn