ros-industrial/fanuc #203: Change of frequency of fanuc driver

Marcofon, 2017-01-18:
Hello, I have a question regarding the Fanuc driver package. I'm following an university project involving an LR 200 robot with a R-30iB Mate controller. The Fanuc tech support said that the controller can reach a loop rate of 125 Hz. In the karel files coming with this package the frequency is set to 40 Hz as default. We tried to change it to 125 but apparently with no luck; in both cases checking the frequency at which the messages are sent with the command rostopic hz /... showed an average rate of 29 Hz. That is even less than the default 40 Hz.

So, is there a way to increase this frequency? Is it something that is controlled by the robot controller and that can't be changed?

Thank you for your attention. If you can, let me know. I can also give you more details if needed or if i wasn't clear enough in my explanation.

gavanderhoorn, 2017-01-19:
Are you asking about the state reporting frequency (ie: `ros_state`), or the 'control frequency' (and I write this in quotes because it really isn't the control frequency, but a throttle on the infinite `while` that deals with network communication and `ros_movesm`)?

In any case: this driver is not suitable for any sort of closed loop control of the robot. It is possible to do typical pick-and-place applications.

Marcofon, 2017-01-23:
Thank you for your reply and sorry for the delay.
We have planned a meeting with the Fanuc Tech Support this week. For this reason I want to reply to this thread when we will have all the informations from the tech support. So I can better reply to your answer.

Thank you very much indeed,

Kind regards

Marco

gavanderhoorn, 2017-01-23:
No problem.

>  The Fanuc tech support said that the controller can reach a loop rate of 125 Hz. 

That is true (with an `ITP` of 8ms you get a 125 Hz interpolation cycle time), but is not very meaningful: for state reporting it is possible to do this at 125 Hz, but without a way to actually *control* the robot at 125 Hz that has limited utility.

Unfortunately Fanuc does not support any kind of high bandwidth, low latency external joint space motion interface, otherwise I would have used it.

Marcofon, 2017-01-26:
After the meeting, we are now able to reach 10ms in reading and 20ms in writing. The Fanuc support is trying to develop some karel programs to help us. I will let you know when we will try all the architecture.

Thank you again for your attention.

gavanderhoorn, 2017-01-26:
Thanks for the update. I'd be interested.

gavanderhoorn, 2017-02-06:
I investigated this a little more. 

The tl;dr version is: task scheduling is done with a fixed time-slice, so all periods must be integer multiples of the time-slice or actual time suspended for a given task will not correspond to what is expected.

---

Slightly longer version: see the following graph (yellow line: desired frequency, magenta: actual frequency, dark blue: ratio actual over desired, x-axis: period in ms):

[image]

The problem with task scheduling on Fanuc controllers is that they use a time-slicing approach. The time-slice is called the *InTerpolation Period* or `ITP`, and its length is configured in the system variable `$SCR.$ITP_TIME`. Typical values are 8 or 12 ms, depending on the type of robot (iirc, LR Mates have 8 ms).

All time related operations on a Fanuc controller are influenced by this ITP (or at least: those that are bound to a Karel interpreter instance, I can't say anything about other controller internals). All timeouts, delays, everything that does something with time is basically limited by the ITP in terms of granularity (or precision, if you will).

Looking at the graph above, the problem with this when dealing with small time intervals (left of the graph) becomes immediately clear. For the case of `DELAY(..)` fi, the smallest timeout value is exactly ITP milliseconds. Anything less than that will be `CEIL`ed up (so 1ms becomes 8, 2 becomes 8, etc).

As for the `ros_state` program not reaching 40 Hz when configured as such: 40 Hz is 25 ms, which is not an integer multiple of 8. The next integer multiple is actually 32 ms, which results in an actual frequency of about 31 Hz. For controllers with a 12 ms ITP, this would mean an actual frequency of 27 Hz.

We have a controller with a 12 ms period here in the lab, and I get exactly 27 Hz of state reporting msgs out of it. The next 'jump' is at ~41 Hz, then at ~83 Hz, which is the maximum frequency our controller can reach, given the inherent limitation due to the way it schedules tasks.

As to why you got 29 Hz when you configured it for 125 Hz I cannot say.

---

Note that this is also documented in the Karel Reference Manual. In a version `G` manual, that would be in table 6-4 *Interval Between Global Condition Handler Scans*. The table shows the relationship between `$COND_TIME` and the desired 'scan time', but it's essentially the same thing (`$COND_TIME` is often equal to `$ITP_TIME`).

gavanderhoorn, 2017-02-06:
@jettan btw.

jettan, 2017-02-06:
Very interesting! As far as I know all our controllers here for the LR Mate as well as the M10 have an ITP of 8ms.

Thanks for the update.

gavanderhoorn, 2017-02-06:
You probably have only R-30iB controllers?

Marcofon, 2017-02-06:
@gavanderhoorn Thank you very much for the update and for the depth investigation, I will inform my supervisor if you already didn't. Thank you!

Marco

gavanderhoorn, 2017-02-06:
@Marcofon: I've just verified that with the current `fanuc_driver` code reaching `1.0/$ITP` Hz state reporting works (and with very little CPU utilisation as well).

I'm now really curious what you did that resulted in such poor performance.

gavanderhoorn, 2017-02-06:
@Marcofon wrote:
> In the karel files coming with this package the frequency is set to 40 Hz as default. We tried to change it to 125 but apparently with no luck; in both cases checking the frequency at which the messages are sent with the command rostopic hz /... showed an average rate of 29 Hz.

just to make sure: did you change the value of `LOOP_HZ` ([here](https://github.com/ros-industrial/fanuc/blob/d00beba883fa5806f9a8cb93d5c2683184d3ce00/fanuc_driver/karel/ros_state.kl#L121)) in the `ros_state.kl` file, or did you change the *config item* (ie: `cfg_.loop_hz`) on the controller?

If the former: did you also recompile, upload and re-init the configuration of `ros_state`?
if the latter: did you restart the `ros_state` program after changing the value?

Without at least one of these two options, nothing is supposed to change, which could explain what you reported.

jettan, 2017-02-08:
@gavanderhoorn: Yeah we only have R30iB controllers here.

Marcofon, 2017-02-13:
Sorry for the late reply @gavanderhoorn . We changed the ros_state.kl and we did everything like you said.

gavanderhoorn, 2017-02-13:
@Marcofon: including removing the old version of `ros_state.pc` and `ros_state.vr` on the controller?

If you did that, then I'm afraid I can't diagnose what went wrong without more info.

Marcofon, 2017-02-13:
Yes, we also did that.

gavanderhoorn, 2017-03-22:
I'm going to close this issue.

State reporting can be done at the maximum rate attainable by the controller (ie: 125 or 83 Hz), and that works with the current code in `fanuc_driver`.

Traj execution / motion relaying at 1/`ITP` is definitely not possible, or at least not without using an option like DPM (and even then),

Without more information on how @Marcofon attained a 40 Hz position update rate, there is nothing for us to do here.

closed 2017-03-22 by gavanderhoorn