ros-industrial/fanuc #207: Tests don't pass on Kinetic

VictorLamoine, 2017-02-16:
The tests in the `fanuc` and `fanuc_experimental` packages don't pass on Ubuntu 16.04 with Kinetic:

```
####
#### Running command: "make cmake_check_build_system" in "/home/im/code/catkin_workspace/build"
####
####
#### Running command: "make test -j8 -l8" in "/home/im/code/catkin_workspace/build"
####
Running tests...
Test project /home/im/code/catkin_workspace/build
      Start  1: _ctest_fanuc_cr35ia_support_roslaunch-check_tests_roslaunch_test.xml
 1/44 Test  #1: _ctest_fanuc_cr35ia_support_roslaunch-check_tests_roslaunch_test.xml ...........................................***Failed    1.94 sec
      Start  2: _ctest_fanuc_lrmate200ib_support_roslaunch-check_tests_roslaunch_test.xml
 2/44 Test  #2: _ctest_fanuc_lrmate200ib_support_roslaunch-check_tests_roslaunch_test.xml ......................................***Failed    2.98 sec
      Start  3: _ctest_fanuc_lrmate200ic5h_moveit_config_roslaunch-check_launch_demo.launch
 3/44 Test  #3: _ctest_fanuc_lrmate200ic5h_moveit_config_roslaunch-check_launch_demo.launch ....................................***Failed    0.86 sec
      Start  4: _ctest_fanuc_lrmate200ic5h_moveit_config_roslaunch-check_tests_moveit_planning_execution.xml
 4/44 Test  #4: _ctest_fanuc_lrmate200ic5h_moveit_config_roslaunch-check_tests_moveit_planning_execution.xml ...................***Failed    1.34 sec
      Start  5: _ctest_fanuc_lrmate200ic5l_moveit_config_roslaunch-check_launch_demo.launch
 5/44 Test  #5: _ctest_fanuc_lrmate200ic5l_moveit_config_roslaunch-check_launch_demo.launch ....................................***Failed    1.01 sec
      Start  6: _ctest_fanuc_lrmate200ic5l_moveit_config_roslaunch-check_tests_moveit_planning_execution.xml
 6/44 Test  #6: _ctest_fanuc_lrmate200ic5l_moveit_config_roslaunch-check_tests_moveit_planning_execution.xml ...................***Failed    1.24 sec
      Start  7: _ctest_fanuc_lrmate200ic_moveit_config_roslaunch-check_launch_demo.launch
 7/44 Test  #7: _ctest_fanuc_lrmate200ic_moveit_config_roslaunch-check_launch_demo.launch ......................................***Failed    0.91 sec
      Start  8: _ctest_fanuc_lrmate200ic_moveit_config_roslaunch-check_tests_moveit_planning_execution.xml
 8/44 Test  #8: _ctest_fanuc_lrmate200ic_moveit_config_roslaunch-check_tests_moveit_planning_execution.xml .....................***Failed    1.04 sec
      Start  9: _ctest_fanuc_lrmate200ic_support_roslaunch-check_tests_roslaunch_test.xml
 9/44 Test  #9: _ctest_fanuc_lrmate200ic_support_roslaunch-check_tests_roslaunch_test.xml ......................................***Failed    6.34 sec
      Start 10: _ctest_fanuc_lrmate200id_support_roslaunch-check_tests_roslaunch_test.xml
10/44 Test #10: _ctest_fanuc_lrmate200id_support_roslaunch-check_tests_roslaunch_test.xml ......................................***Failed    2.08 sec
      Start 11: _ctest_fanuc_m10ia_moveit_config_roslaunch-check_launch_demo.launch
11/44 Test #11: _ctest_fanuc_m10ia_moveit_config_roslaunch-check_launch_demo.launch ............................................***Failed    0.85 sec
      Start 12: _ctest_fanuc_m10ia_moveit_config_roslaunch-check_tests_moveit_planning_execution.xml
12/44 Test #12: _ctest_fanuc_m10ia_moveit_config_roslaunch-check_tests_moveit_planning_execution.xml ...........................***Failed    1.03 sec
      Start 13: _ctest_fanuc_m10ia_support_roslaunch-check_tests_roslaunch_test_m10ia.xml
13/44 Test #13: _ctest_fanuc_m10ia_support_roslaunch-check_tests_roslaunch_test_m10ia.xml ......................................***Failed    2.07 sec
      Start 14: _ctest_fanuc_m10ia_support_roslaunch-check_tests_roslaunch_test_m10ia7l.xml
14/44 Test #14: _ctest_fanuc_m10ia_support_roslaunch-check_tests_roslaunch_test_m10ia7l.xml ....................................***Failed    1.88 sec
      Start 15: _ctest_fanuc_m16ib20_moveit_config_roslaunch-check_launch_demo.launch
15/44 Test #15: _ctest_fanuc_m16ib20_moveit_config_roslaunch-check_launch_demo.launch ..........................................***Failed    1.03 sec
      Start 16: _ctest_fanuc_m16ib20_moveit_config_roslaunch-check_tests_moveit_planning_execution.xml
16/44 Test #16: _ctest_fanuc_m16ib20_moveit_config_roslaunch-check_tests_moveit_planning_execution.xml .........................***Failed    1.13 sec
      Start 17: _ctest_fanuc_m16ib_support_roslaunch-check_tests_roslaunch_test.xml
17/44 Test #17: _ctest_fanuc_m16ib_support_roslaunch-check_tests_roslaunch_test.xml ............................................***Failed    1.82 sec
      Start 18: _ctest_fanuc_m20ia10l_moveit_config_roslaunch-check_launch_demo.launch
18/44 Test #18: _ctest_fanuc_m20ia10l_moveit_config_roslaunch-check_launch_demo.launch .........................................***Failed    0.87 sec
      Start 19: _ctest_fanuc_m20ia10l_moveit_config_roslaunch-check_tests_moveit_planning_execution.xml
19/44 Test #19: _ctest_fanuc_m20ia10l_moveit_config_roslaunch-check_tests_moveit_planning_execution.xml ........................***Failed    1.30 sec
      Start 20: _ctest_fanuc_m20ia_moveit_config_roslaunch-check_launch_demo.launch
20/44 Test #20: _ctest_fanuc_m20ia_moveit_config_roslaunch-check_launch_demo.launch ............................................***Failed    0.81 sec
      Start 21: _ctest_fanuc_m20ia_moveit_config_roslaunch-check_tests_moveit_planning_execution.xml
21/44 Test #21: _ctest_fanuc_m20ia_moveit_config_roslaunch-check_tests_moveit_planning_execution.xml ...........................***Failed    1.18 sec
      Start 22: _ctest_fanuc_m20ia_support_roslaunch-check_tests_roslaunch_test.xml
22/44 Test #22: _ctest_fanuc_m20ia_support_roslaunch-check_tests_roslaunch_test.xml ............................................***Failed    3.09 sec
      Start 23: _ctest_fanuc_m430ia2f_moveit_config_roslaunch-check_launch_demo.launch
23/44 Test #23: _ctest_fanuc_m430ia2f_moveit_config_roslaunch-check_launch_demo.launch .........................................***Failed    0.96 sec
      Start 24: _ctest_fanuc_m430ia2f_moveit_config_roslaunch-check_tests_moveit_planning_execution.xml
24/44 Test #24: _ctest_fanuc_m430ia2f_moveit_config_roslaunch-check_tests_moveit_planning_execution.xml ........................***Failed    1.18 sec
      Start 25: _ctest_fanuc_m430ia2p_moveit_config_roslaunch-check_launch_demo.launch
25/44 Test #25: _ctest_fanuc_m430ia2p_moveit_config_roslaunch-check_launch_demo.launch .........................................***Failed    0.82 sec
      Start 26: _ctest_fanuc_m430ia2p_moveit_config_roslaunch-check_tests_moveit_planning_execution.xml
26/44 Test #26: _ctest_fanuc_m430ia2p_moveit_config_roslaunch-check_tests_moveit_planning_execution.xml ........................***Failed    1.16 sec
      Start 27: _ctest_fanuc_m430ia_support_roslaunch-check_tests_roslaunch_test.xml
27/44 Test #27: _ctest_fanuc_m430ia_support_roslaunch-check_tests_roslaunch_test.xml ...........................................***Failed    3.14 sec
      Start 28: _ctest_fanuc_m6ib_support_roslaunch-check_tests_roslaunch_test.xml
28/44 Test #28: _ctest_fanuc_m6ib_support_roslaunch-check_tests_roslaunch_test.xml .............................................***Failed    1.90 sec
      Start 29: _ctest_fanuc_m710ic_additive_manufacturing_moveit_config_roslaunch-check_launch_demo.launch
29/44 Test #29: _ctest_fanuc_m710ic_additive_manufacturing_moveit_config_roslaunch-check_launch_demo.launch ....................***Failed    1.33 sec
      Start 30: _ctest_fanuc_m710ic_additive_manufacturing_moveit_config_roslaunch-check_tests_moveit_planning_execution.xml
[... 53 lines cut ...]
	 19 - _ctest_fanuc_m20ia10l_moveit_config_roslaunch-check_tests_moveit_planning_execution.xml (Failed)
	 20 - _ctest_fanuc_m20ia_moveit_config_roslaunch-check_launch_demo.launch (Failed)
	 21 - _ctest_fanuc_m20ia_moveit_config_roslaunch-check_tests_moveit_planning_execution.xml (Failed)
	 22 - _ctest_fanuc_m20ia_support_roslaunch-check_tests_roslaunch_test.xml (Failed)
	 23 - _ctest_fanuc_m430ia2f_moveit_config_roslaunch-check_launch_demo.launch (Failed)
	 24 - _ctest_fanuc_m430ia2f_moveit_config_roslaunch-check_tests_moveit_planning_execution.xml (Failed)
	 25 - _ctest_fanuc_m430ia2p_moveit_config_roslaunch-check_launch_demo.launch (Failed)
	 26 - _ctest_fanuc_m430ia2p_moveit_config_roslaunch-check_tests_moveit_planning_execution.xml (Failed)
	 27 - _ctest_fanuc_m430ia_support_roslaunch-check_tests_roslaunch_test.xml (Failed)
[... cut ...]
	 33 - _ctest_fanuc_m900ib_support_roslaunch-check_tests_roslaunch_test.xml (Failed)
	 34 - _ctest_fanuc_r1000ia_support_roslaunch-check_tests_roslaunch_test.xml (Failed)
	 41 - _ctest_fanuc_driver_roslaunch-check_tests_roslaunch_test.xml (Failed)
	 42 - _ctest_industrial_robot_simulator_roslaunch-check_launch (Failed)
	 44 - _ctest_fanuc_m710ic_additive_manufacturing_support_roslaunch-check_tests_roslaunch_test.xml (Failed)
Errors while running CTest
Makefile:127 : la recette pour la cible « test » a échouée
make: *** [test] Erreur 8
Invoking "make test -j8 -l8" failed

```

gavanderhoorn, 2017-02-16:
Thanks.

But without seeing the output of one of the failed tests it's hard to say what could be the cause :)

VictorLamoine, 2017-02-16:
It fails to find any node:

```bash
####
#### Running command: "make cmake_check_build_system" in "/home/victor/catkin_workspace/build"
####
####
#### Running command: "make run_tests_fanuc_m10ia_support -j8 -l8" in "/home/victor/catkin_workspace/build"
####
Scanning dependencies of target run_tests_fanuc_m10ia_support_roslaunch-check_tests_roslaunch_test_m10ia.xml
Scanning dependencies of target run_tests_fanuc_m10ia_support_roslaunch-check_tests_roslaunch_test_m10ia7l.xml
-- run_tests.py: execute commands
-- run_tests.py: execute commands
  /usr/bin/cmake -E make_directory /home/victor/catkin_workspace/build/test_results/fanuc_m10ia_support
  /usr/bin/cmake -E make_directory /home/victor/catkin_workspace/build/test_results/fanuc_m10ia_support
  /opt/ros/kinetic/share/roslaunch/cmake/../scripts/roslaunch-check -o /home/victor/catkin_workspace/build/test_results/fanuc_m10ia_support/roslaunch-check_tests_roslaunch_test_m10ia7l.xml.xml /home/victor/catkin_workspace/src/fanuc/fanuc_m10ia_support/tests/roslaunch_test_m10ia7l.xml 
  /opt/ros/kinetic/share/roslaunch/cmake/../scripts/roslaunch-check -o /home/victor/catkin_workspace/build/test_results/fanuc_m10ia_support/roslaunch-check_tests_roslaunch_test_m10ia.xml.xml /home/victor/catkin_workspace/src/fanuc/fanuc_m10ia_support/tests/roslaunch_test_m10ia.xml 
checking /home/victor/catkin_workspace/src/fanuc/fanuc_m10ia_support/tests/roslaunch_test_m10ia.xml
checking /home/victor/catkin_workspace/src/fanuc/fanuc_m10ia_support/tests/roslaunch_test_m10ia7l.xml
...writing test results to /home/victor/catkin_workspace/build/test_results/fanuc_m10ia_support/roslaunch-check_tests_roslaunch_test_m10ia7l.xml.xml
FAILURE:
[/home/victor/catkin_workspace/src/fanuc/fanuc_m10ia_support/tests/roslaunch_test_m10ia7l.xml]:
	cannot find node [joint_trajectory_action] in package [industrial_robot_client]
cannot find node [robot_state_bswap] in package [fanuc_driver]
cannot find node [robot_state] in package [fanuc_driver]
cannot find node [motion_streaming_interface_bswap] in package [fanuc_driver]
cannot find node [motion_streaming_interface] in package [fanuc_driver]
wrote test file to [/home/victor/catkin_workspace/build/test_results/fanuc_m10ia_support/roslaunch-check_tests_roslaunch_test_m10ia7l.xml.xml]
...writing test results to /home/victor/catkin_workspace/build/test_results/fanuc_m10ia_support/roslaunch-check_tests_roslaunch_test_m10ia.xml.xml
FAILURE:
[/home/victor/catkin_workspace/src/fanuc/fanuc_m10ia_support/tests/roslaunch_test_m10ia.xml]:
	cannot find node [joint_trajectory_action] in package [industrial_robot_client]
cannot find node [robot_state_bswap] in package [fanuc_driver]
cannot find node [robot_state] in package [fanuc_driver]
cannot find node [motion_streaming_interface_bswap] in package [fanuc_driver]
cannot find node [motion_streaming_interface] in package [fanuc_driver]
wrote test file to [/home/victor/catkin_workspace/build/test_results/fanuc_m10ia_support/roslaunch-check_tests_roslaunch_test_m10ia.xml.xml]
-- run_tests.py: verify result "/home/victor/catkin_workspace/build/test_results/fanuc_m10ia_support/roslaunch-check_tests_roslaunch_test_m10ia7l.xml.xml"
Built target run_tests_fanuc_m10ia_support_roslaunch-check_tests_roslaunch_test_m10ia7l.xml
-- run_tests.py: verify result "/home/victor/catkin_workspace/build/test_results/fanuc_m10ia_support/roslaunch-check_tests_roslaunch_test_m10ia.xml.xml"
Built target run_tests_fanuc_m10ia_support_roslaunch-check_tests_roslaunch_test_m10ia.xml

```

gavanderhoorn, 2017-02-16:
That seems strange. I would almost wonder if the right `setup.bash` has been `source`d?

VictorLamoine, 2017-02-16:
It seems like the related nodes haven't been built at all.
find . -iname "robot_state_bswap" returns nothing.

gavanderhoorn, 2017-02-16:
Did you first build the workspace?

You can't just run the test target without building the packages. Missing targets will not trigger a build.

gavanderhoorn, 2017-02-16:
See [roslaunch_add_file_check(..) and build dependencies](http://answers.ros.org/question/125676) on ROS Answers fi.

VictorLamoine, 2017-02-16:
Yep that's the problem; there is no dependency between the `roslaunch_add_file_check` and the nodes so you have to build the package before runing these tests

VictorLamoine, 2017-02-16:
Right on, I'll read that right now, I did not know about this limitation!

gavanderhoorn, 2017-02-16:
So to clarify that answer: you can specify direct dependencies, but that doesn't really help if your dependencies have dependencies (which is almost always the case).

gavanderhoorn, 2017-02-16:
See also #185 (and #187) for why the `DEPENDENCIES` were removed.

VictorLamoine, 2017-02-16:
Makes sense, so there's no perfect solution at the moment.
I'll trigger a build on GitLab CI before running the tests then!

gavanderhoorn, 2017-02-16:
I'll close this one as `works-for-me` for now.

Re-open if you feel there is a need.

closed 2017-02-16 by gavanderhoorn