ros-industrial/fanuc #225: Error: 'Connection refused' (errno: 111)

cvanclee, 2017-07-25:
I've been having trouble connecting to our robot over the network here, so I decided to use a direct connection from the robot to my laptop. However, I haven't been able to get the direct connection to work either. Is there some configuration problem with our robot or my ethernet connection? I'll post some of my info, and if you need any more, please ask.

I'm running
`roslaunch fanuc_lrmate200id_moveit_config moveit_planning_execution.launch sim:=false robot_ip:=10.0.0.93`
Which produces the error
`[ERROR] [1500991666.939646322]: Failed to connect to server, rc: -1. Error: 'Connection refused' (errno: 111)`

Pinging the robot succeeds 100% of the time.
Here is the ethernet information
[image]

gavanderhoorn, 2017-07-25:
First thing to check would be whether the Karel programs on the controller are actually running.

Can you tell me the *exact* steps you take when starting everything on the controller?

cvanclee, 2017-07-25:
-Jog robot into spot I can reach the head
-Add/remove tool
-Set coordinates to USER
-Flip switch on table to PROGRAM
-Flip switch in top left of controller to the left
-Press reset to remove any errors
-Press Select on controller, navigate to 93 ROS, press Enter
-Press green switch on table
-Run `roslaunch fanuc_lrmate200id_moveit_config moveit_planning_execution.launch sim:=false robot_ip:=10.0.0.93`
Hopefully I didn't miss anything

gavanderhoorn, 2017-07-25:
Ok, that may have been a bit too exact - I don't think I have a 'table' here with a switch on it :), I also don't know what *Flip switch in top left of controller to the left* means. Are you talking about the mode selector switch (T1/T2/Auto)?

In any case, please open http://10.0.0.93/MD/PRGSTATE.DG, and check the programs listed there. `ROS_STATE`, `ROS_TRAJ` and `ROS` should all be in the `RUNNING` state. If they are not, that could be the cause of what you're seeing.

If it is, could you please try to start the `ROS` TP program in Manual/T2 mode (you'll have to keep the deadman and shift pressed in that mode)?

---

Your issue reminds me of gavanderhoorn/fanuc_driver_exp#2 btw. Please see if it is related.

gavanderhoorn, 2017-07-25:
> Set coordinates to USER

btw: I don't know if you do that for the ROS driver, but it's not going to affect anything on the ROS side.

cvanclee, 2017-07-25:
Ok, so I started up the ROS program and then checked http://10.0.0.93/MD/PRGSTATE.DG. I was unable to find a `ROS_TRAJ` by ctrl-fing. I didn't run anything from my laptop, I just started the program from the controller. Is that why? Here are the other two parameters you mentioned:

```
6          ROS RUNNING @ 169 in SSOCK_ACCEPT of LIBSSOCK

******  History Data  ******
Routine depth: 3  Routine: SSOCK_ACCEPT                        
Line:   169       Program: LIBSSOCK                              Type: PC    

Routine depth: 2  Routine: SSOCK_ACCPT2                        
Line:   254       Program: LIBSSOCK                              Type: PC    

Routine depth: 1  Routine: ROS_RELAY                           
Line:   247       Program: ROS_RELAY                             Type: PC    

Routine depth: 0  Routine: ROS                                 
Line:     3       Program: ROS                                   Type: TP    

7    ROS_STATE RUNNING @ 169 in SSOCK_ACCEPT of LIBSSOCK

******  History Data  ******
Routine depth: 2  Routine: SSOCK_ACCEPT                        
Line:   169       Program: LIBSSOCK                              Type: PC    

Routine depth: 1  Routine: SSOCK_ACCPT2                        
Line:   254       Program: LIBSSOCK                              Type: PC    

Routine depth: 0  Routine: ROS_STATE                           
Line:   216       Program: ROS_STATE                             Type: PC    
```

> (note that if you haven't already, I would recommend you migrate to fanuc_driver_exp)
What do you mean by this? I haven't seen anything about migrating to that.

> If it is, could you please try to start the ROS TP program in Manual/T2 mode (you'll have to keep the deadman and shift pressed in that mode)?
I couldn't find `ROS_TRAJ`, so should I hold off on trying this?

cvanclee, 2017-07-25:
> I also don't know what Flip switch in top left of controller to the left means

I'm not sure what to call the switch or the positions, so here's a picture. I'm assuming the one I have it set at is auto, and the other is manual
[image]

gavanderhoorn, 2017-07-25:
That is the Teach Pendant, not the controller.

And it is the TP ON/OFF switch.

gavanderhoorn, 2017-07-25:
Ignore `ROS_TRAJ`, I was under the impression you had already migrated to [gavanderhoorn/fanuc_driver_exp](https://github.com/gavanderhoorn/fanuc_driver_exp). It's a better version of the driver, fully bw-compatible.

`ROS_TRAJ` is `ROS_RELAY` in `fanuc_driver`.

But something is not right, as this is what I see when I run the `ROS` TP program:

```
     ROS_RELAY RUNNING @ 169 in SSOCK_ACCEPT of LIBSSOCK

******  History Data  ******
Routine depth: 2  Routine: SSOCK_ACCEPT
Line:   169       Program: LIBSSOCK               Type: PC

Routine depth: 1  Routine: SSOCK_ACCPT2           
Line:   254       Program: LIBSSOCK               Type: PC

Routine depth: 0  Routine: ROS_RELAY              
Line:   263       Program: ROS_RELAY              Type: PC

           ROS RUNNING @ 12 in ROS_MOVESM of ROS_MOVESM

******  History Data  ******
Routine depth: 1  Routine: ROS_MOVESM             
Line:    12       Program: ROS_MOVESM             Type: TP

Routine depth: 0  Routine: ROS                    
Line:     3       Program: ROS                    Type: TP

     ROS_STATE RUNNING @ 169 in SSOCK_ACCEPT of LIBSSOCK

******  History Data  ******
Routine depth: 2  Routine: SSOCK_ACCEPT           
Line:   169       Program: LIBSSOCK               Type: PC

Routine depth: 1  Routine: SSOCK_ACCPT2           
Line:   254       Program: LIBSSOCK               Type: PC

Routine depth: 0  Routine: ROS_STATE              
Line:   231       Program: ROS_STATE              Type: PC
```

in your output, I see `ROS` as a direct parent of `ROS_RELAY`, while in my output, I see `ROS` as a stand-alone program, waiting in `ROS_MOVESM`, and `ROS_STATE` and `ROS_RELAY` also in `RUNNING` state, but as separate tasks.

That is also what I would expect, as both `ROS_STATE` and `ROS_RELAY` are started with a `RUN` TP statement, while `ROS_MOVESM` is `CALL`ed.

Have you changed anything to any of the involved (TP) programs? Can you compare the `ROS` program on your TP with the source ([here](https://github.com/ros-industrial/fanuc/blob/248f3acd3f607ad0c4d66b596694345891df85a7/fanuc_driver/tpe/ros.ls#L22-L24))?

gavanderhoorn, 2017-07-25:
I'm starting to wonder whether this is not the same issue that the OP in gavanderhoorn/fanuc_driver_exp#2 ran into.

Has the driver ever worked on your setup? Have you always started it this way? Or have you only recently started running things in AUTO?

gavanderhoorn, 2017-07-25:
Just to make sure things are running: after you've started things on the controller, try these commands. If they don't error out with a "can't connect" error, at least that bit is ok.

```
nc 10.0.0.93 11000
```

and

```
nc 10.0.0.93 11002
```

If either of these doesn't work, try starting `ROS` in Manual-T1 mode. If it works, then it would seem to be similar to gavanderhoorn/fanuc_driver_exp#2.

cvanclee, 2017-07-25:
> Have you changed anything to any of the involved (TP) programs? Can you compare the ROS program on your TP with the source (here)?

How do I view the ROS source on my TP? I can't find it.

> Has the driver ever worked on your setup? Have you always started it this way? Or have you only recently started running things in AUTO?

I'm fairly new here, but as far as I know we haven't worked with ROS before. We've been running the robot with TP and pc programs loaded from USB.

I may not have have time to try the rest of your suggestions today, so I'll get back to you on the rest tomorrow.

gavanderhoorn, 2017-07-25:
@cvanclee wrote:
> How do I view the ROS source on my TP? I can't find it.

`SELECT` -> `TYPE` -> `2 TP PROGRAMS`. Scroll down to `ROS`, press `ENTER`.

You should now see the contents of the TP program.

cvanclee, 2017-07-25:
That brings up a screen with this

1: RUN ROS_STATE
2: RUN ROS_MOVESM
3: CALL ROS_RELAY
[End]

gavanderhoorn, 2017-07-25:
When did you install the driver? The `ROS` program was updated to:

```
RUN ROS_STATE
RUN ROS_RELAY
CALL ROS_MOVESM
```

in December 2014 (248f3acd3f607ad0c4d66b596694345891df85a7).

What is the comment on the `ROS` TP program? Is it `r2` or `r3`? And all the other programs (both TP and Karel)?

cvanclee, 2017-07-25:
All the programs that have an `r` are `r2`, including `ROS`. I don't see any `r3`s

cvanclee, 2017-07-25:
I believe we have had the robot since at least 2014, and no one ever bothered updating it. Doing that and your other suggestions will be the first thing I do tomorrow. Thank you for the excellent help!

cvanclee, 2017-07-26:
> nc 10.0.0.93 11000
> 
> and
> 
> nc 10.0.0.93 11002

I tried these commands, but I'm not sure what they should do to be considered 'working'. `nc 10.0.0.93 11000` seems to just run without output, I have to ctrl-c to get my terminal prompt. `nc 10.0.0.93 11002` prints out a stream of strange symbols

gavanderhoorn, 2017-07-26:
And how did you start the `ROS` TP program this time? Manually or with the cycle start button?

The `11000` command is supposed to not print anything, the `11002` should. So it sounds like things are ok on the controller. At least, this time.

cvanclee, 2017-07-26:
I started it with the cycle start button, which is how I've always done it

gavanderhoorn, 2017-07-26:
Then you should be able to connect to it with the `fanuc_driver` nodes.

If not, then the issue is not on the controller side, but on the ROS side / your pc.

gavanderhoorn, 2017-07-26:
I would still recommend you migrate / upgrade to `fanuc_driver_exp` btw.

If you could contact me via email, I would like to ask you to test something for me which should make installation of the driver much easier and faster.

cvanclee, 2017-07-26:
> Then you should be able to connect to it with the fanuc_driver nodes.

How do I test this?

> If not, then the issue is not on the controller side, but on the ROS side / your pc.

That was my original thought, but I have no idea what the problem could be. I tried it with another laptop and got the same error.

> If you could contact me via email, I would like to ask you to test something for me which should make installation of the driver much easier and faster.

I would be happy to once I get permission to update the drivers

gavanderhoorn, 2017-07-26:
@cvanclee wrote:
> > Then you should be able to connect to it with the fanuc_driver nodes.
> 
> How do I test this?

```
roslaunch fanuc_lrmate200id_support robot_state_visualize_lrmate200id.launch robot_ip:=10.0.0.93
```

That should show you RViz with your robot.

If that doesn't complain, try:

```
roslaunch fanuc_lrmate200id_support robot_interface_streaming_lrmate200id.launch robot_ip:=10.0.0.93
```

This won't start RViz, but if it can't connect, you'll get errors.

>> If not, then the issue is not on the controller side, but on the ROS side / your pc.
> 
> That was my original thought, but I have no idea what the problem could be. I tried it with another laptop and got the same error.

Well as I wrote earlier, starting things in AUTO is not as straightforward and can result in weird issues. See gavanderhoorn/fanuc_driver_exp#2 I referenced earlier.

>> If you could contact me via email, I would like to ask you to test something for me which should make installation of the driver much easier and faster.
>
> I would be happy to once I get permission to update the drivers

Perhaps good to know is that the new driver is fully bw-compatible with the old one, and only requires updating the controller side.

And if you have one, you could first make an *image backup* of your robot onto a PCMCIA/CF-card. If the upgrade doesn't work or there is something not ok, you can revert back to the exact state your robot was in before you did the upgrade.

I would recommend making image backups of your controller regularly anyway, so perhaps investing in a PCMCIA C(ompact) F(lash) converter would be good. They should be about $20 on Amazon.

cvanclee, 2017-07-26:
I ran the commands you gave in AUTO mode both from our network and from a direct link. From the network, both executed correctly (the network connects to the robot, but only about 50% of the time and will drop the connection eventually, which is why I'm here trying a direct connection). From a direct connection, the first command opens RViz without the robot model and continuously prints 
```
[ERROR] [1501076136.096786364]: Failed to connect to server, rc: -1. Error: 'Network is unreachable' (errno: 101)
[ERROR] [1501076136.096881469]: Failed to receive message length
[ERROR] [1501076136.096936947]: Failed to receive incoming message
```

cvanclee, 2017-07-26:
Scratch that, my ethernet connection settings had the wrong subnet mask on it. Both commands execute correctly from a direct connection as well

gavanderhoorn, 2017-07-26:
One thing you could try to see if we can diagnose the network side of this is to run Wireshark on the same machine that is running ROS.

Install Wireshark 2 (from the PPA) and [packet-simplemessage](https://github.com/ros-industrial/packet-simplemessage). Then start a capture and do what you always do.

If it works, retry (as we're not interested in that).

As soon as you have a capture of a 'not working' run, save it, compress it and send it to me -- or make it available in some other way.

gavanderhoorn, 2017-07-26:
I looked at the wireshark capture you sent me: afaict, the ports just really aren't open. I really have the impression that the two Karel programs just don't get started correctly, which causes the network init to fail.

In the capture it's `ROS_STATE` that doesn't seem to have come up properly.

I would really recommend you try starting the `ROS` TP program in Manual/T1 mode (so keep both deadman and `shift` pressed), then start the ROS side (with your other hand). If I'm correct, this should work 10/10 times.

If this fixes it, you'll have to start looking into what the other user did in gavanderhoorn/fanuc_driver_exp#2 to get things to start properly in Auto mode.

And I'm really curious if this ever worked correctly. Perhaps you can ask your colleagues / team mates about that.

cvanclee, 2017-07-27:
Sorry, but how do I start ROS on the TP in manual mode? The cycle start doesn't work

gavanderhoorn, 2017-07-27:
`SELECT` -> `TYPE` -> `2 TP PROGRAMS`. Scroll down to `ROS`, press `ENTER`.

Switch the controller to T1 or T2 mode using the mode selector switch.

Turn the TP *ON*, press the deadman, press `RESET`. Everything should be green now.

Press and hold down `SHIFT`, now press `FWD`.

Keep holding the deadman and `SHIFT`.

gavanderhoorn, 2017-07-27:
But really, this is *working with Fanuc robots 101*. Have you not received any instruction before starting to work with the robot?

gavanderhoorn, 2017-07-27:
And **please** make sure that your cell, your robot and you are safe before following the instructions I wrote two comments up.

If in doubt, as your co-workers.

cvanclee, 2017-07-27:
Ok, I got it started. The robot loaded into the correct position in RViz (matching the real robot), but then an error keeps printing out.
```
[ERROR] [1501158351.122026904]: Failed to connect to server, rc: -1. Error: 'Transport endpoint is already connected' (errno: 106)
[ERROR] [1501158351.122177795]: Failed to receive message length
[ERROR] [1501158351.122236058]: Failed to receive incoming message
```

So I did a second start, this time the robot does not load into the correct position (it loads into all zeros) and this error keeps printing:
```
[ERROR] [1501158560.035874929]: Failed to connect to server, rc: -1. Error: 'Connection refused' (errno: 111)
[ERROR] [1501158560.035984688]: Failed to receive message length
[ERROR] [1501158560.036019426]: Failed to receive incoming message
```

> But really, this is working with Fanuc robots 101. Have you not received any instruction before starting to work with the robot?

Well yes, just mostly in relation to getting it to run with ROS and a few other basic functions

cvanclee, 2017-07-27:
Should I still give this a shot?

> In the System/Config menu you need to set
> 
>     option 43 Remote/Local setup: Local (or $REMOTE_CFG.$REMOTE_TYPE = OFF)
>     option 56 No motion PR operate mode to TRUE.
> 
> In the Prog Select menu you need to set:
> 
>     option 1 Program select mode: OTHER and DETAIL to ROS ($SHELL_WRK.$CUST_NAME)
>     option 2 Production start method: UOP

From the issue you linked me

gavanderhoorn, 2017-07-27:
Tbh I'm having a hard time diagnosing this. Especially the fact that things sometimes work, and other times don't, or only intermittently is complicating matters. The distance is another factor.

And I'm starting to suspect the hw in your controller at this point. Getting those errors after a successful connect seems to point to a dropped connection (that's typically when the 'Transport endpoint is already connected' pops up). I can trigger this by removing the ethernet cable from the controller without shutting down the programs and then reconnecting the controller to the network.

Could you please make some more wireshark captures so we can see what is happening?

Do you have any other networked programs on your controller you can test?

cvanclee, 2017-07-27:
> Could you please make some more wireshark captures so we can see what is happening?

Sent a couple

> Do you have any other networked programs on your controller you can test?

None that we know of besides ROS

> Edit: and you don't see any output on the TP when you get those errors on the ROS side?

I've seen RSTA jnt-pkt-srlise err: -67211 the past few runs

> Edit2: I'm assuming this is an R-30iA controller. I'm a little confused by the screenshot of your NIC status in your OP: it says 1000mb/s, while the R-30iA has 100BASE-TX ports (ie: 100mb/s max).

It is a R-30iB, but I thought that was referring to the ethernet cable itself

gavanderhoorn, 2017-07-27:
>> Could you please make some more wireshark captures so we can see what is happening?
>
> Sent a couple

Looking at them I see rather a lot of *spurious retransmissions*, *missing segment* and other TCP related problems I don't expect to see. There are also instances where the controller sends multiple simple message msgs in one TCP frame, as if it has had to wait before it could transfer data again. That is also not normal.

>> Edit: and you don't see any output on the TP when you get those errors on the ROS side?
> 
> I've seen RSTA jnt-pkt-srlise err: -67211 the past few runs

`67-211` is `HOST-211`, which is "SM: Connection Write Buffer is full". That would seem to indicate that the remote side (ie: ROS) is not reading incoming data fast enough. That would be strange, but not impossible. Another possibility is that the connection has already been lost, but the ctrlr (with a socket timeout of 1 minute) hasn't noticed yet.

>> Edit2: I'm assuming this is an R-30iA controller. I'm a little confused by the screenshot of your NIC status in your OP: it says 1000mb/s, while the R-30iA has 100BASE-TX ports (ie: 100mb/s max).
> 
> It is a R-30iB, but I thought that 1000mb/s was referring to the ethernet cable itself

I was under the impression that the screenshot showed the status when you had a direct connection.

An R-30iB also has only a 100MBit/s NIC, so the 1000Mb/s screenshot must have been with a switch in between you and the controller.

---

Two things:

 1. can you check `/var/log/syslog` on your Linux machine and see whether there are messages related to your NIC? Look for things like 'negotiating' or 'dropping' or 'missed' or similar wording.
 1.  we're going to take ROS out of the picture: start wireshark, start the `ROS` TP program, then in a terminal, run `nc 10.0.0.93 11002 > /dev/null`. Leave that running. Look at wireshark. See whether you start to see a lot of red-on-black lines. If `nc` drops the connection (ie: returns to the terminal) that is also not good.

cvanclee, 2017-07-27:
> I was under the impression that the screenshot showed the status when you had a direct connection.
> 
> An R-30iB also has only a 100MBit/s NIC (at least, that is what the docs tell me), so the 1000Mb/s screenshot must have been with a switch in between you and the controller.

Unless I'm extremely mistaken, it was a direct connection between the controller and my laptop. The 1000Mb/s is referring to the maximum load the ethernet cable can handle, I think.

1. I didn't see many of the keywords you mentioned in my syslog, and the ones I did see didn't seem related. I could send you the file?
2. `nc` did not drop the connection, but wireshark has a lot of bad looking packets from my ip. I'll send that file in a moment

gavanderhoorn, 2017-07-27:
> Unless I'm extremely mistaken, it was a direct connection between the controller and my laptop. The 1000Mb/s is referring to the maximum load the ethernet cable can handle, I think.

No. The `1000Mb/s` is the actual *link speed* of the selected NIC. If that was a direct connection, then it would suggest your NIC had negotiated (assuming you haven't disabled auto-negotiation) a `1000MBit/s` link with the controller. That should be impossible (as the ctrlr should not be capable of that afaict).

cvanclee, 2017-07-27:
Well, I have no idea what to make of it then.

gavanderhoorn, 2017-07-27:
Would be nice if you could verify it by setting up another direct connection between your ROS PC and the controller. It's probably not the cause of the problem here, but would clear up at least something.

cvanclee, 2017-07-27:
Ok, this was done with my wifi switch turned off, with an ethernet cable from the controller to my laptop, the only ethernet connection on either my laptop or controller is this cable. I am able to ping the robot:
```
cvancleef@cvancleef-Latitude-E6410:~$ ping 10.0.0.93
PING 10.0.0.93 (10.0.0.93) 56(84) bytes of data.
64 bytes from 10.0.0.93: icmp_seq=1 ttl=60 time=1.63 ms
```

And... now it shows 100mb/s. I have no clue why it said 1000 before. So in case something else changed, I tried to run some commands on RViz with the direct connection, but still get connection refused, same as before.

gavanderhoorn, 2017-07-27:
Ok. With the direct connection at 100mb/s, do the `nc` test again. Check wireshark.

gavanderhoorn, 2017-07-27:
Could you post the ethernet statistics from http://10.0.0.93/MD/ETHERNET.DG?

gavanderhoorn, 2017-07-27:
Afaict from the wireshark traces, your network or your NIC or the NIC in the controller is dropping packets. That leads to everything backing up with the `HOST-211` errors as a result.

gavanderhoorn, 2017-07-27:
One more thing to test: start a wireshark capture, and connect to the controller with an FTP client (filezilla or something), open the `md:` device and download everything there to some temp folder on your PC. Check wireshark and look for the red-on-black packets.

For a normal / correctly functioning network, I would expect almost no *dup acks* or other problems.

If there are then try disabling auto-negotation for the direct link (using something like [this](https://www.shellhacks.com/change-speed-duplex-ethernet-card-linux/)) and see if that helps. Set the connection to 100Mbit, full-duplex.

Check ethernet statistics on the controller again.

And please do everything with the `ROS` TP program started *manually* in T1 or T2 mode.

---

If after this there are still problems than I think the problem is really either with your laptop or with the controller hardware / software and I don't think I can help you.

gavanderhoorn, 2017-07-27:
In your latest capture ("nc with direct connection, wifi turned off" https://github.com/ros-industrial/fanuc/issues/225#issuecomment-318405372) I don't see any dups or other problems. Are you still not able to (reliably) connect?

Is your wifi network using the same IP range as the wired one? If so, and when you weren't using the direct connection *and* had not turned off your wireless, it could very well be that Linux was still routing connections to your robot over your wireless NIC. That could explain the dropped packets and other problems.

cvanclee, 2017-07-27:
> In your latest capture ("nc with direct connection, wifi turned off" #225 (comment)) I don't see any dups or other problems. Are you still not able to (reliably) connect?

I direct connected for about 10 minutes without dropping the connection, so it may be fixed. I have no idea what has changed to make it work, but oh well. I'm going to leave it alone for a while and then do a direct connect again make sure it still works.

> One more thing to test: start a wireshark capture, and connect to the controller with an FTP client

I'm using filezilla with the robot and my laptop back on the network and put 10.0.0.49 for Host, and my log in information for username and password. I get
```
Status:	Connecting to 10.0.0.49:21...
Status:	Connection attempt failed with "ECONNREFUSED - Connection refused by server".
Error:	Could not connect to server
```

gavanderhoorn, 2017-07-27:
> I direct connected for about 10 minutes without dropping the connection, so it may be fixed. I have no idea what has changed to make it work, but oh well

I'd rather not make this an "oh well" issue. I'd rather know what was going on.

Looking back through your wireshark captures I'm pretty sure that you were connecting over wireless the entire time. Packets from your laptop appear to be coming from a LiteOn/HP/Dell/OEM card (OUI `70:f1:a1`). That is a wireless device. The "direct connection" capture uses a Dell device (`00:26:b9`).

That would explain the missing packets and lost connections.

As I wrote above, even if you connect a wire, Linux may still route your traffic over your wireless connection (as it could have a higher *metric*).

gavanderhoorn, 2017-07-27:
As to the FTP connection: you are trying to connect to your own laptop, not to the robot controller. Afaik, the controller is at `10.0.0.93`, not `.49`.

But I believe this isn't necessary any more, as we probably found the cause of the problems.

gavanderhoorn, 2017-07-27:
TCP/IP is very much capable of dealing with lossy links, but not in a way that makes it useful for us. In addition, the TCP/IP stack on the Fanuc controller isn't very great - or at least, our interface to it. Lost connections or dropped packets will make it choke, as you found out.

Let's hope your wired connection works.

If it does, could I ask you to confirm and then close the issue?

gavanderhoorn, 2017-07-27:
And for (my) future reference: this was basically still ros-industrial/fanuc_experimental#32.

cvanclee, 2017-07-27:
So it looks like I have to disconnect my wireless because it takes precedence over a direct connection. Is there a way to have the direct connection take precedence for when I launch ROS so I can run it while still having internet access? Should I make a separate question about this?

closed 2017-07-27 by cvanclee

gavanderhoorn, 2017-07-27:
That is not a ROS related question, but a general Linux OS / distro (ie: Ubuntu) one. It's a system configuration task.

I should have paid more attention when looking at the wireshark captures. That could've saved us a lot of time.

gavanderhoorn, 2017-07-27:
I would still recommend you upgrade the ROS-I driver parts on the controller to `fanuc_driver_exp`, but that is obviously up to you.

cvanclee, 2017-07-27:
Well, it looks like I made a couple of oversights as well, so I don't hold it against you. Thanks for the help again!