ros-industrial/fanuc #227: How to connect fanuc_driver to Fanuc controller?

hustfzt2, 2017-09-26:
I am trying to connect fanuc_driver to the controller using the IP address, however, it fails. 
Can anyone give the detailed procedure about the connection between fanuc_driver and controller like the motoman_ROS

gavanderhoorn, 2017-09-26:
Please provide us with more information, saying "it fails" is not enough.

Did you follow all [available tutorials](http://wiki.ros.org/fanuc/Tutorials)?

If you did and something doesn't work, describe the exact steps you used and what you observed, otherwise we cannot help you.

hustfzt2, 2017-09-27:
Thank you for your help, Now I have conenected fanuc_driver with robotguide software successfully.

By jogging the joint buttons in robotguide, I can see the synchronous action in RVIZ and robotguide, however,  by sending the ros topic /joint_state in ros node, it has little influence on robotguide.

In addition, when i try to connect fanuc_driver with real robot, user socket message (USM) is not available in Fanuc teach pendent.

gavanderhoorn, 2017-09-27:
@hustfzt2 wrote:
> By jogging the joint buttons in robotguide, I can see the synchronous action in RVIZ and robotguide,

ok, good. State reporting works in that case.

> however, by sending the ros topic /joint_state in ros node, it has little influence on robotguide.

That is expected, as the driver exposes a [FollowJointTrajectory](http://docs.ros.org/jade/api/control_msgs/html/action/FollowJointTrajectory.html) action interface. It does not listen to `JointState` msgs. The `/joint_state` topic is for reporting current state of the robot, not for controlling it.

> In addition, when i try to connect fanuc_driver with real robot, user socket message (USM) is not available in Fanuc teach pendent.

Are you encountering an `INTP-320: (LIBSSOCK, N) Undefined built-in`? If so, see [wiki/fanuc/Troubleshooting: INTP-320: (LIBSSOCK, N) Undefined built-in](http://wiki.ros.org/fanuc_driver/Troubleshooting#INTP-320:_.28LIBSSOCK.2C_N.29_Undefined_built-in).

The `User Socket Messaging` option (`R648`) is a required option I'm afraid. If your robot doesn't have that, `fanuc_driver` is not going to work.

guoquheweilai, 2017-09-27:
If the controller doesn't have R632, we can continue, but USM is
absolutely necessary.

gavanderhoorn, 2017-09-27:
Correct.

gavanderhoorn, 2017-09-28:
Seeing as your original question was answered I'm going to close this issue.

If you have any further questions please post a new one or re-open if it's appropriate.

closed 2017-09-28 by gavanderhoorn

JerrryHe, 2019-10-31:
> If the controller doesn't have R632, we can continue, but USM is
> absolutely necessary.
@gavanderhoorn  @guoquheweilai 
I was wondering how to use fanuc_driver if there is no KAREL(R32).  Could you tell me the details, thanks.