ros-industrial/fanuc #236: Import R-2000iC/125L support pkg from A5 project

gavanderhoorn, 2018-02-16:
Part of the 'reach study' pkg, but removed in a later commit.

See [fanuc_r2000ic_125l_support](https://github.com/ros-industrial-consortium/A5/tree/1c515dc21845c4f744388d706a6718fa6c308d07/a5_reach_study/robot_models/fanuc_r2000ic_125l_support).

Should go into `fanuc_experimental`.

gavanderhoorn, 2018-02-16:
@marip8: is that pkg ok as-is, or would it need additional work before adding it to `fanuc_experimental`?

marip8, 2018-02-16:
I believe the only thing the package is missing is the actual models for the robot (which I was unable to find at the time I created the package) and probably a few clean-up/documentation details. If you could point me to where I could get the CAD models for the R2000, I would be happy to make a pull request with the URDF

gavanderhoorn, 2018-02-16:
So from which robot are the meshes in the `meshes` directory then?

https://github.com/ros-industrial-consortium/A5/blob/1c515dc21845c4f744388d706a6718fa6c308d07/a5_reach_study/robot_models/fanuc_r2000ic_125l_support/meshes/r2000ic_125l/visual/link_3.stl

seems like a proper mesh to me?

marip8, 2018-02-16:
Those are actually the links from the Motoman MH180-120, which was pretty close to the same size and layout. I haven't been able to find anything more than datasheets for the R2000

gavanderhoorn, 2018-02-16:
Ah, ok. You haven't asked Fanuc? That would be where you'd get the meshes from typically.

gavanderhoorn, 2018-02-16:
I only have meshes for the R-2000iB/125L. Not sure whether there are any differences with the `iC`.

gavanderhoorn, 2019-10-08:
If/when we get a request for R-2000iC/125L we can revisit this.

I have an R-2000iC/165F at the moment, not the variant discussed in the OP.

Closing.

closed 2019-10-08 by gavanderhoorn

simonschmeisser, 2019-11-21:
@gavanderhoorn I have a project with a R-2000iC/165F upcoming, would you mind opening a PR? Thanks a lot!