ros-industrial/fanuc #249: Build errors on Ubuntu 18.04 / ROS Melodic

MrDadaGuy, 2019-02-12:
Unable to build fanuc in ROS industrial -- lots of errors.  I've tried to suppress deprecated warnings but still many, many errors.

The output is too large to paste here, so I've put it on pastebin:

https://pastebin.com/QQSeAije

Other packages from ros industrial are working fine, so I'm not sure what's going on with the fanuc packages.  I do have a particular interest in getting fanuc working. Thanks

VictorLamoine, 2019-02-12:
Duplicate of https://github.com/ros-industrial/fanuc/issues/241
(how to fix is provided in the issue)

gavanderhoorn, 2019-02-12:
Yes, this would seem to be a duplicate of #241.

@MrDadaGuy: the IKFast plugins (hosted by the `fanuc_X_moveit_plugins` packages) are not compatible with Melodic at the moment. They probably never will be, unless people really want that and community patches are provided.

See #241 (as already mentioned) and #245 for some discussion on an alternative.

gavanderhoorn, 2019-02-12:
As this is a duplicate, I'm closing this.

closed 2019-02-12 by gavanderhoorn

MrDadaGuy, 2019-02-12:
Thanks y'all -- I had seen that issue but I did not make the connection.  Applying the suggested fix from #241 did work, and now I can build!  Thanks very much!