ros-industrial/fanuc #258: Trying to compile X_moveit_plugins on melodic

joao-pm-santos96, 2019-03-09:
When compiling on ubutnu 18.04, ROS Melodic, it can't find the function 'getTipFrame()' that it defined on 'kinematics_base.h' that I verified to have in my system. Is there come way to solve it? 

The package includes moveit_core but the path to 'kinematics_base.h': is /opt/ros/melodic/include/moveit/kinematics_base

[image]

@miguelriemoliveira

gavanderhoorn, 2019-03-09:
This is a duplicate of #241.

Not that it isn't a problem, but let's keep the nr of duplicates to a minimum.

closed 2019-03-09 by gavanderhoorn

miguelriemoliveira, 2019-03-09:
Hi @joao-pm-santos96 

I will try to compile on my laptop. 

Can you create a section in this repository's readme file in which you explain how to install / build this package (whatever steps you have done until now)?  

For now the issue is assigned to you ... Then I will try to compile from my side.

miguelriemoliveira, 2019-03-09:
@joao-pm-santos96 , please add me as a collaborator to your repository ...

miguelriemoliveira, 2019-03-09:
and I think you had an answer already ... great !

gavanderhoorn, 2019-03-09:
I'm slightly confused because of the many comments here ..

miguelriemoliveira, 2019-03-09:
Sorry @gavanderhoorn , my fault. I thought I was writting on @joao-pm-santos96 (my students) repository. 
I did not realize it was another repository. I will try to remove my comments.

gavanderhoorn, 2019-03-09:
@joao-pm-santos96: please don't post screenshots of terminals. It's all text, so copy-paste the console text into your comments.

joao-pm-santos96, 2019-03-09:
@gavanderhoorn sorry, I'm new to github but I'm very thankful for your quick answer it solved the problem!
@miguelriemoliveira Yeah I didn't add it to my repository, nevertheless I'll add you as contributor, also thank you for the quick reply.