ros-industrial/fanuc #265: Fanuc M20iA35 Config

cpetersmeier, 2019-05-22:
Hello everyone,

I am working in an industrial setting with **Fanuc M20iA / 35kg Robots**.
I am in the process of adapting the existing nodes for other versions of the robot to the 35 kg variant. 
I was wondering if this work has already been done by someone, or if this has potential to be added to the whole package.

Chris

VictorLamoine, 2019-05-22:
I've not already done this robot!

gavanderhoorn, 2019-05-22:
I'm not aware of anyone adding support for the `/35M` variant.

> I am in the process of adapting the existing nodes for other versions of the robot to the 35 kg variant.

which nodes are you referring to specifically here? The driver and most other software is completely manipulator-agnostic, so it should suffice to create a new xacro and add some launch files.

cpetersmeier, 2019-05-22:
> The driver and most other software is completely manipulator-agnostic, so it should suffice to create a new xacro and add some launch files.

That is great to hear! 

I just looked at an overview about the ros-i packages and started some demos. I haven't looked into the specific nodes yet.

I'll start with xacro from some 3D-meshes of the robot and try to go from there.

gavanderhoorn, 2019-05-22:
As this is not an issue with the packages in this repository, I'm going to close the issue.

Feel free to keep commenting on the issue of course @cpetersmeier.

closed 2019-05-22 by gavanderhoorn

cpetersmeier, 2019-05-24:
@gavanderhoorn 
I forked and adapted the package to the best of my knowledge.

I still have a problem in that the kinematics are not loading when I start the demo launch nodes and moveit commander node.

Shall I open a pull request for some review?

gavanderhoorn, 2019-05-24:
From https://github.com/cpetersmeier/fanuc/commit/1254e956476cd9e99e1f7ef4362db1152e5eedf9 I get the impression you're trying to edit an IKFast solver to work with your robot.

Don't do that: it's too error prone and most likely won't work.

IKFast solvers are auto-generated code. They're not meant to be edited manually.

Either generate a new solver, or use something like [trac_ik](http://wiki.ros.org/trac_ik).

> Shall I open a pull request for some review?

For now I can probably take a look at [your fork](https://github.com/cpetersmeier/fanuc). No need for a PR yet.