ros-industrial/fanuc #266: joint_state_publisher not starting

cpetersmeier, 2019-06-13:
Hello, 

I am unable to use the moveit commander cmd line, because the joint_state_publisher doesnt start properly.
In my launch file I call
`
<node name="joint_state_publisher" pkg="joint_state_publisher" type="joint_state_publisher"> 
<param name="use_gui" value="true"/>
<rosparam param="source_list">[move_group/fake_controller_joint_states]</rosparam>
</node>
`

When executing the launch file I get this error:

_
Traceback (most recent call last):
  File "/opt/ros/melodic/lib/joint_state_publisher/joint_state_publisher", line 474, in <module>
    jsp = JointStatePublisher()
  File "/opt/ros/melodic/lib/joint_state_publisher/joint_state_publisher", line 149, in __init__
    robot = xml.dom.minidom.parseString(description)
  File "/usr/lib/python2.7/xml/dom/minidom.py", line 1928, in parseString
    return expatbuilder.parseString(string)
  File "/usr/lib/python2.7/xml/dom/expatbuilder.py", line 940, in parseString
    return builder.parseString(string)
  File "/usr/lib/python2.7/xml/dom/expatbuilder.py", line 223, in parseString
    parser.Parse(string, True)
UnicodeEncodeError: 'ascii' codec can't encode character u'\xb0' in position 4420: ordinal not in range(128)
_

and

_
[joint_state_publisher-4] process has died [pid 5986, exit code 1, cmd /opt/ros/melodic/lib/joint_state_publisher/joint_state_publisher __name:=joint_state_publisher __log:=/home/chris/.ros/log/a08dc708-8dca-11e9-9608-000c2908badd/joint_state_publisher-4.log].
log file: /home/chris/.ros/log/a08dc708-8dca-11e9-9608-000c2908badd/joint_state_publisher-4*.log
_

This is all in an rviz context, and the gui and the rest of moveit loads fine.
Can anyone help me resolve this issue?

gavanderhoorn, 2019-06-13:
I'm unsure how this is related to the Fanuc packages here.

Can you clarify?

gavanderhoorn, 2019-06-13:
And please use verbatim formatting for xml and console copy-pastes (ie: three consecutive backticks).

cpetersmeier, 2019-06-13:
Of course, 
so I got this error while trying to launch the test.launch file provided by the fanuc packages

gavanderhoorn, 2019-06-13:
> so I got this error while trying to launch the test.launch file provided by the fanuc packages

please be specific: *which* file.

Provide steps to reproduce, otherwise we cannot help you.

cpetersmeier, 2019-06-13:
Okay, I am sorry for my unclear explanation. 

I am working on adapting the fanuc packages for the 35kg variant of the m20ia robot.

What I did:

Create a URDF / XACRO file for the robot.
Create a MoveIt Config.
Created the 4 launch files, described [here](http://wiki.ros.org/Industrial/Tutorials/WorkingWithRosIndustrialRobotSupportPackages)

**When do I get the error?**
When launching the...
- adapted test_m20ia.launch file (note: appears to not load the origin and material tags correctly)
- demo.launch file created by the moveIt setup assistant (note: loads correctly)

When starting the test_m20ia.launch file from the offical ros-i fanuc packages, everything works fine.

I am wondering if I missed a step in my launch file, or moveit configuration. The other nodes, such as the moveGroup and Planning Scene provided by moveIt start just fine. That is why I thought it would maybe be an issue with a thing I missed in adapting the packages.
I followed the syntax and naming schemes of the offical packages as close as possible.

gavanderhoorn, 2019-06-13:
It's difficult to help you without seeing your files, but I would first check whether you're not running into locale issues, seeing as you're German, using Ubuntu Bionic/ROS Melodic and most likely using a German locale.

Refer to https://github.com/ros/urdfdom_headers/issues/45.

cpetersmeier, 2019-06-13:
Thanks for the hint! I'll try and dig deeper there.
When I run 
`$ locale`
the output I get is
`LANG=en_US.UTF-8`
`LANGUAGE=`
`LC_CTYPE="en_US.UTF-8"`
`LC_NUMERIC="en_US.UTF-8"`
`LC_TIME="en_US.UTF-8"`
`LC_COLLATE="en_US.UTF-8"`
`LC_MONETARY="en_US.UTF-8"`
`LC_MESSAGES="en_US.UTF-8"`
`LC_PAPER="en_US.UTF-8"`
`LC_NAME="en_US.UTF-8"`
`LC_ADDRESS="en_US.UTF-8"`
`LC_TELEPHONE="en_US.UTF-8"`
`LC_MEASUREMENT="en_US.UTF-8"`
`LC_IDENTIFICATION="en_US.UTF-8"`
`LC_ALL=`

gavanderhoorn, 2019-06-13:
That looks ok.

The problem is most likely this:

```
UnicodeEncodeError: 'ascii' codec can't encode character u'\xb0' in position 4420: ordinal not in__range(128)
```

`\xb0` is (according to [this site](http://www.fileformat.info/info/unicode/char/b0/index.htm)) a *Unicode Character 'DEGREE SIGN'*.

It's likely that you copy-pasted joint limits from a PDF and added a degree sign somewhere.

You'll have to check your xacro macro.

gavanderhoorn, 2019-06-13:
As this is not a problem with any of the packages in this repository, I'm going to close the issue.

Feel free to keep commenting on it though, @cpetersmeier.

closed 2019-06-13 by gavanderhoorn

cpetersmeier, 2019-06-13:
That solved the issue! Thank you very much for the fast help! 

For reference, here is a detailled description of the solution:

In the .XACRO file I commented the joint limits for reference with a degree sign:
`<limit lower="-4.712389" upper="4.712389" effort="0" velocity="6.98" /> `
``

Removing the degree sign in the comments solved the issue for me:
``

gavanderhoorn, 2019-06-13:
I would recommend to use the support in `xacro` for Python expressions.

That would allow you to do something like this:

```xml
<limit lower="${radians(-450)}" upper="${radians(450)}" effort="0" velocity="..." />
```