ros-industrial/fanuc #267: Arcmate series support

pvlvsk, 2019-06-14:
Hello community,

is there support for Arcmate Series (100ic 12) planned in the future?

Thanks!

gavanderhoorn, 2019-06-14:
That depends a bit on what you by "support".

Last time I checked, Arcmates are essentially M-10s, M-20s and LR Mates, but with Arctool and support for controlling welding power sources.

Both `fanuc_driver` and `fanuc_driver_exp` are completely manipulator agnostic, so there is no need to adapt it to any robot type.

The ROS side could use the support packages and moveit configurations of the regular robot models and variants, so I believe it should work out-of-the-box as long as the regular model is already supported.

`fanuc_m10ia_support` currently does not support the `/12` variant, so that would have to be added.

I don't have plans right now to do that. A PR adding that variant would certainly be welcomed.

gavanderhoorn, 2019-06-14:
As this is more a question than an actionable issue, I'm closing this.

Feel free to keep commenting of course.

closed 2019-06-14 by gavanderhoorn