ros-industrial/fanuc #282: driver: Karel Move command on R-30iB+

asierfernandez, 2019-10-23:
FYI, in case it is interesting for someone.

We've been playing with Fanuc CR-7iA/L collaborative robot with R-30iB+ controller.
We though Karel `Move` command was deprecated, also commented in #86, but we tested in a simple Karel program listening desired joint_state from sockets and passing them to `Move` command. The robot moves! Additionally defining `$TERMTYPE=NODECEL` we can get a kind of a smooth motion.

gavanderhoorn, 2019-10-23:
Thanks for your report.

> We though Karel `Move` command was deprecated

It certainly is, but deprecated doesn't mean it doesn't function any more.

Unfortunately we only have Fanuc's "word" for this: they've indicated (to me, and others) on multiple occasions that `MOVE TO` is not to be used any more. Especially for robots with multiple groups using it could lead to "dangerous and unexpected movement" of robots. In addition, the "motion planner behind Karel has not been updated for many years" and "will not be able to reach optimal robot performance".

And according to some sources, use of `MOVE TO` may also lead to Fanuc engineers becoming unwilling to support your application/deployment, at least on recent controllers (but that is hear-say).

With this in mind, we've removed it from the list of candidate approaches.

gavanderhoorn, 2019-10-23:
See https://github.com/ros-industrial/fanuc/compare/indigo-devel...gavanderhoorn:karel_moveto_experiment for an earlier experiment I've done with `MOVE TO` and `fanuc_driver` btw.

gavanderhoorn, 2019-10-23:
As this is not an issue with the package(s) in this repository, I'll close it.

Feel free to keep commenting of course.

closed 2019-10-23 by gavanderhoorn

asierfernandez, 2019-10-23:
Thanks for your explanation. We'll take it into account!

aaronkui, 2022-11-11:
amazing