ros-industrial/fanuc #285: stop() function of moveit is performing randomly

jediofgever, 2020-02-10:
Hi, 

I am developing a desktop GUI to control my robot, there is essential functionalities and one of these is `stop` button. While testing this GUI with robot, I found out that the `stop()` function performs very randomly, sometimes it stops robot, sometimes it does not and sometimes it stops it with delay. I don't know if this is related with moveit or here, has anyone encountered this ? 
Stop functionality is very crucial for my application , any help is appreciated 
Thank you

gavanderhoorn, 2020-02-10:
You don't tell us which driver you are using, but:

 - if you're using `fanuc_driver`: #93
 - if you're using `fanuc_driver_exp`: gavanderhoorn/fanuc_driver_exp#3

These are known issues.

See the issues for some discussion.

gavanderhoorn, 2020-02-10:
As we already have issues for this, I'm going to close this one.

Please continue discussion on the linked issues.

gavanderhoorn, 2020-02-10:
And @jediofgever: could you please only post issues on the `fanuc_experimental` tracker when those issues are about packages hosted by that repository? Everything else should be posted on either `fanuc` or `fanuc_driver_exp`.

closed 2020-02-10 by gavanderhoorn

jediofgever, 2020-02-10:
Hi @gavanderhoorn,
my bad and Thank you,
I use  `fanuc_driver_exp`
I am going to see the issue there