ros-industrial/fanuc #286: Document supported variants of a model (M-10iA)

simonschmeisser, 2020-02-13:
I'm currently looking at a Fanuc M-10iA/12 and wondering if that matches the generic m10ia here. I will figure it out by comparing drawings to urdf/mesh/opw parameters but it would be nice to explicitely spell out the matching variants.

There seem to exist the following variants:

M-10iA/7L -> m10ia7l
M-10iA/8L ->  -
M-10iA/10M -> ?
M-10iA/10MS -> -
M-10iA/12 -> m10ia ?
M-10iA/12S -> -

https://www.fanuc.eu/pl/en/robots/robot-filter-page/m-10-series

I can then open a PR against the readme.md?

gavanderhoorn, 2020-02-13:
If a variant is not mentioned, it's not supported.

Compare the manifest for the LR Mate 200iC:

https://github.com/ros-industrial/fanuc/blob/63e4bd4b484ed4a0e0f8dfd582333211b5919014/fanuc_lrmate200ic_support/package.xml#L10-L15

For the M-10iA, we have the following variants:

https://github.com/ros-industrial/fanuc/blob/63e4bd4b484ed4a0e0f8dfd582333211b5919014/fanuc_m10ia_support/package.xml#L10-L19

Afaik there is no correspondance to any other variant, unless you're thinking of the Paint and/or Welding robots.

simonschmeisser, 2020-02-13:
> there used to be a "plain" M-10iA. That would be the variant supported by that model.

That seems to be the source of my confusion, sorry. Your model seems to be from 2009, ours is from 2014.

>I don't know whether that would correspond to the /12 variant these days.

According to https://www.fanuc.co.jp/en/product/catalog/pdf/robot/RM-10iAN(E)-02.pdf
at least the kinematics match so possibly it was just a rebranding. I'll try to find some material (mesh/picture) about wether the collision contours have changed or not.

Thanks a lot!

gavanderhoorn, 2020-02-25:
Is this something you'd like to keep open @simonschmeisser ?

simonschmeisser, 2020-02-26:
Hmm, I'm not sure. It was not obvious for me that there is a older "plain" model but it's not a real issue.

closed 2020-02-26 by simonschmeisser

gavanderhoorn, 2020-02-26:
>  I'll try to find some material (mesh/picture) about wether the collision contours have changed or not.

Did you get a chance to verify this?

simonschmeisser, 2020-07-24:
@axelschroth I keep forgetting to ask you ... but it's in production now so I guess the answer is yes

gavanderhoorn, 2020-07-25:
So did the collision meshes change, or did they not?

axelschroth, 2020-07-27:
I would say, that the meshes did not change. So we can use the collision meshes for the /12 variant. But I cannot tell for sure, because I did not measure the geometries and there is no high chance of collision in this particular setup anyway.