ros-industrial/fanuc #288: Can fanuc_M_moveit_config  work with Roboguide?

honghongzhenbang, 2020-04-03:
In tutorials I find moveit can work with real robot or industrial_robot_simulator, no Roboguide. And I start the ros on the TP in Roboguide, On Linux PC, run:
`roslaunch fanuc_m10ia_moveit_config moveit_planning_execution.launch sim:=false use_bswap:=false robot_ip:=xxxx`
Then the TP show connected and moveit can get the robot joint state from Roboguide, but when execute plan show failed to connect to server.
So ,Can fanuc_M_moveit_config  work with Roboguide?

Thank you!

gavanderhoorn, 2020-04-03:
Yes, you certainly can use Roboguide as a stand-in for a real controller. I use this almost weekly.

> And I start the ros on the TP in Roboguide

How exactly do you start everything, and *what* do you start exactly?

honghongzhenbang, 2020-04-04:
@gavanderhoorn Thanks for your reply. I ran the rosstate TPE program on TP, then finish it and ran the ros TPE program without Enter, so the actual program executed is rosstate.  Now I have successfully solved the problem.

closed 2020-04-04 by honghongzhenbang

gavanderhoorn, 2020-04-04:
Only running `rosstate` would indeed cause the problem you reported.

Note btw: you may want to use [fanuc_driver_exp](https://github.com/gavanderhoorn/fanuc_driver_exp) instead of `fanuc_driver`.

It's still not going to be great, but it's better than `fanuc_driver`.

LiperLee, 2025-04-28:
When running roslaunch fanuc_m10ia_moveit_config moveit_planning_execution.launch sim:=false use_bswap:=false robot_ip:=xxxx, How to solve the problem of being unable to execute?

gavanderhoorn, 2025-05-01:
@LiperLee: please create a new post in the [Discussion](https://github.com/ros-industrial/fanuc/discussions) forum.

This is a really old issue here and already closed, so not a good place to debug your problem.