ros-industrial/fanuc #290: Fanuc M710iC-50 kinematics

christianlandgraf, 2020-06-04:
Hello,

while working with the Fanuc M710ic50 and fanuc_m710ic_support package, I noticed that the robot behaviour differs from the real robot and the layout in Visual Components. It seems, as if there is a mistake in the urdf of this package.

For example, if I set the following joint states as can be seen in the picture, the robot is in the displayed position:
[image]

But if I set the same joint states in Visual Components, it looks like:
[image]

Even without the exact solution one cane see, that this is not correct.

Since the actual robot behaves as Visual Component expects, I suppose there is a bug in the ROS package. I will have a look and try to fix it but maybe there is someone else out there, who is more experienced with URDF and can immediately fix the bug. 

I can also give the robot parameters which are set in Visual Components, if desired. Unfortunateley, I do not have access to the real robot at the moment.

Thanks a lot
Christian

gavanderhoorn, 2020-06-04:
I suspect Visual Components is showing you the joint angles as the teach pendant would, which includes the so called "joint 2 - joint 3 coupling".

In ROS, we do not do this, and give you the plain joint angles.

Please ros-industrial/fanuc_experimental#17 for a similar case.

gavanderhoorn, 2020-06-04:
And see also [J3 on TP and joint_3 in ROS do not match](http://wiki.ros.org/action/show/fanuc_driver/Troubleshooting#J3_on_TP_and_joint_3_in_ROS_do_not_match) in the Troubleshooting guide.

gavanderhoorn, 2020-06-04:
I'm going to assume the linked issue and the FAQ entry explain the cause of what you're seeing.

Assuming that's true, I'm going to close this issue, as it's not really a problem with the robot support package -- or at least not something we can fix: "joint coupling" is not a concept in ROS.

If it turns out there is actually something wrong, we can re-open.

In the meantime, feel free to keep commenting of course.

closed 2020-06-04 by gavanderhoorn

christianlandgraf, 2020-06-04:
Thanks a lot for the quick answer! I wasn't aware of that, it worked with the Troubleshooting guide.