ros-industrial/fanuc #298: Fanuc on ROS2

jan-kacina, 2020-12-10:
Hi guys,
I would like to control Fanuc robot using ROS2, mainly not only due to better performace but mainly fue unsupported Python required by ROS1.

Do you plan to implement Fanuc package also for ROS2? Or in more general scope, will be ROS Industrial upgraded for ROS2? If not in near future, have you tried how ROS1 Bridge works with Fanuc?

Thanks and regards,
Jan

gavanderhoorn, 2020-12-11:
Porting the driver to ROS 2 will not result in better performance. The bottleneck is not on the ROS side, but on the Fanuc side (more specifically: in the way the Karel and TP code is written).

Replacing ROS 1 with ROS 2 will not influence this.

> mainly due to unsupported Python required by ROS1.

ROS Noetic supports Python `3.8` compatible. Is that not sufficient?

> Do you plan to implement Fanuc package also for ROS2? 

Not right now.

Ideally, I'd like to do away with anything `simple_message` based, and `fanuc_driver` and `fanuc_driver_exp` are `simple_message` based.

> Or in more general scope, will be ROS Industrial upgraded for ROS2? 

I can't comment on that here in this repository.

> If not in near future, have you tried how ROS1 Bridge works with Fanuc?

I've not tried it, but seeing as the only thing you do is send `FollowJointTrajectory` goals at a very low frequency (I would almost say *ad-hoc*), I don't expect there to be much problems. You would need to use something like https://github.com/ros2/ros1_bridge/pull/256 though, as the standard bridge does not support actions natively.

gavanderhoorn, 2020-12-11:
As this is more of a question than an actionable issue, I'm going to close it.

Feel free to keep commenting on it of course.

closed 2020-12-11 by gavanderhoorn

jan-kacina, 2020-12-14:
Hi @gavanderhoorn,
thanks for the quick response.
Just a few clarifications missing in my original post...

- by performance I meant ROS performance, which should be better in ROS2 (based on pure information only, not an own experience)

- Python 3.8 is indeed OK; I had out-of-date information about Python 2.x supported by ROS. Thanks for correction.

Just one question remains... Based on [this table](http://repositories.ros.org/status_page/compare_kinetic_melodic_noetic.html), Fanuc packages are available only for Kinetic. Will it be upgraded for Noetic? Or can I use Kinetic packages somehow in Noetic?

Thanks.

gavanderhoorn, 2020-12-14:
You should be able to build the packages in a `catkin_ws` with Noetic as well.

See the [Building](https://github.com/ros-industrial/fanuc#building) section in the readme.