ros-industrial/fanuc #310: CR7iA Joint Limit Different from the RoboGuide

brianFruit, 2021-02-06:
The CR7iA xacro is showing a different joint limit, so I went ahead and update the xacro file. However, I still can't move the robot to outside of the original limits. For example, I tried both move_group.move and move_group.execute to this joint configuration, joint 1: -0.0621894 | joint 2: 1.43175 | **joint 3: -1.4553** | joint 4: 0.0171653 | joint 5: 0.711367 | joint 6: -1.19186, but the ROS side is complaining with the errors. I think the collision check is preventing it to move, but I don't know how to fix it.

I'm running it on Ubuntu 18, ROS Melodic. 

[image]

[image]

gavanderhoorn, 2021-02-07:
The CR-7iA support package was contributed in ros-industrial/fanuc_experimental#35 by a user. I did not have access to any documentation on the model at the time, so we had to rely on his information.

That PR was contributed almost 5 years ago, so there is a good chance that Fanuc has changed the joint limits for the CR-7iA (and `/L`) since that time. They have released several revisions of both the documentation and system software for their collaborative models.

Looking at the data from the system variables, the limits for all axes are (current day Roboguide):

```yaml
lowerlims:
  - -2.967060e+00
  - -1.151917e+00
  - -2.262121e+00
  - -3.316125e+00
  - -2.094395e+00
  - -6.283185e+00
upperlims:
  - 2.967060e+00
  - 1.745329e+00
  - 4.258603e+00
  - 3.316125e+00
  - 2.094395e+00
  - 6.283185e+00
```

Compared to the limits in the `xacro:macro` it does look like `joint_3` is incorrect.

> I think the collision check is preventing it to move, but I don't know how to fix it.

it could be that the collision mesh is too coarse and restricts motion unnecessarily. See ros-industrial/fanuc_experimental#34 for an earlier similar issue with the base LR Mate 200iD.

> Update:
> I disabled the collision check by adding the tag below into the srdf, but it crashed the ROS side and disconnected it from the RoboGuide.
> <disable_collisions link1="link_2" link2="link_4" reason="UserCustom" />

the driver does not use this information at all, so I'm not sure how a crash in MoveIt (which I'm just going to assume it was) would affect us here.

Without the actual error message (copy-pasted as text, please don't post screenshots of terminals) it's going to be difficult to say anything about this.

gavanderhoorn, 2021-02-07:
Please also note: Fanuc does not specify the upper and lower position limits separately in their documentation for the CR series of robots. Nor do they list any maximum joint speeds. Only an overall Cartesian maximum velocity is specified.

I'm not sure, but I would assume they do this as the (extensive) safety features of the robot will never allow it to use the maximum joint capabilities anyway in collaborative mode.

As we don't model Cartesian limits in URDFs, nor do we have access to the safety/collaborative controller of Fanuc, we can only spec whatever joint limits the system variables give us, but that does not mean the actual robot will be able to execute the trajectories you plan with those.

gavanderhoorn, 2021-02-07:
I've corrected the limits in #311.

I've also added a note about joint limits vs the collaborative safety system used with the CR series of robots.

As to the collision mesh, that would be a different issue.

gavanderhoorn, 2021-02-07:
I'm closing this, as the joint limits should now be correct -- or at least: correspond to what Roboguide shows you on the TP.

As to the collision mesh, please post a new issue if that's still a problem.

closed 2021-02-07 by gavanderhoorn

brianFruit, 2021-02-08:
Thanks for the help! I will post the collision mesh error separately.

commit 2aa58011 (2021-02-07): cr7ia: fix joint limits. Fix #310.

Based on the system variables, the limits for various joints were incorrect.

It could be Fanuc has changed the limits since the initial contribution of the support package.