ros-industrial/fanuc #315: How to load fanuc model(eg.m710ic50.xacro) in gazebo?

zhaoyu19920930, 2021-05-14:
Hi, I am making simulation on gazebo with the model m710ic50.xacro, here is my launch file:
`<launch>
<arg name="limited" default="false" doc="If true, limits joint range [-PI, PI] on all joints."/>
<arg name="paused" default="false" doc="Starts gazebo in paused mode"/>
<arg name="gui" default="true" doc="Starts gazebo gui"/>

<include file="$(find gazebo_ros)/launch/empty_world.launch">
<arg name="world_name" default="worlds/empty.world"/>
<arg name="paused" value="$(arg paused)"/>
<arg name="gui" value="$(arg gui)"/>
</include>
<param name="robot_description" command="$(find xacro)/xacro --inorder '$(find fanuc_m710ic_support)/urdf/m710ic50.xacro'"/>
<node name="spawn_gazebo_model" pkg="gazebo_ros" type="spawn_model" args="-urdf -param robot_description -model fanuc_m710ic50 -z 0.2" respawn="false" output="screen"/>
</launch>`

But the model can not load in the gazebo, I find the m710ic50.xacro has no inertial and mass parameters.

My confusions:
(1) Can I make simulation on gazebo with  m710ic50.xacro model？
(2) If yes, how to configure the inertial and mass？Can you show some examples?

Hope to receive your response soon, Thanks.

gavanderhoorn, 2021-05-14:
> Can I make simulation on gazebo with m710ic50.xacro model？

not without adding `inertial` and `dynamics` elements to `m710ic50_macro.xacro`.

> If yes, how to configure the inertial and mass？Can you show some examples?

you'll have to either get that information from Fanuc, find literature on system identification performed on that robot, or use estimates.

There is a related PR (https://github.com/ros-industrial/fanuc_experimental/pull/20) which you could perhaps use as a template to see how to add Gazebo support for a particular model/variant.

The best way to do it is to separate Gazebo support as much as possible, and introduce a `fanuc_m710ic_gazebo` package. The referenced PR should give you an idea how you could do that.

gavanderhoorn, 2021-05-14:
We used to track adding Gazebo support in #145. However it was deemed too much work which could not be justified (demand seems low).

As that situation hasn't really changed, I'm going to close this issue as well.

@zhaoyu19920930: feel free to keep commenting of course.

We can always re-open.

closed 2021-05-14 by gavanderhoorn