ros-industrial/fanuc #316: CRX10 drivers and modules

Juxi, 2021-06-15:
We have a potential project where we plan on using a FANUC CRX10, newer cobot variants. I can’t find information in the package about its compatibility with Ros (and Ros-industrial).
I was wondering if there is some experience/knowledge about working with that manipulator. 

We have cad files, and we can create urdfs out of these if that helps.

gavanderhoorn, 2021-06-15:
All the drivers we have are robot and controller agnostic, so as long as there is still support for Karel and TP, things should start moving. I haven't checked personally, as I haven't had any time on a CRX or the newer controllers. But all it takes is starting up a roboguide simulation and following the regular installation tutorials of the Karel + TP programs.

You won't be able to exploit any of the "cobot features", but that would be a problem with any external form of control of the CRX I believe.

> We have cad files, and we can create urdfs out of these if that helps.

A support package would indeed need the meshes and a `xacro:macro`. A `xacro:macro` is not that complex, and starting from something like the M-10iA support package should make the basic structure clear. It would then be a matter of updating `link` lengths, `joint` limits and the `mesh`es.

Juxi, 2021-06-18:
Great thanks, I will try to get the roboguide sim setup and let you know how it goes.

gavanderhoorn, 2021-06-18:
There appears to be a user on ROS Answers who's created the support package already: [Add a ROS package for an unsupported Fanuc robot model](https://answers.ros.org/question/370229). Perhaps you could comment you can test it (also)?

Zedderrs, 2021-06-20:
https://github.com/ros-industrial/fanuc/pull/317 aims to implement the crx10ial

gavanderhoorn, 2022-03-31:
With the merge of ros-industrial/fanuc_experimental#67, there is now a basic support package for the CRX-10iA/L.

Even though that doesn't include the base CRX-10iA/L, I'm going to close this issue, as the non-`/L` would be "just another variant" and should not be difficult to add.

closed 2022-03-31 by gavanderhoorn