ros-industrial/fanuc #325: ROS Driver on Fanuc R-J3iB?

marabount, 2021-07-04:
Hello,

I would like to test the ROS driver on the Fanuc M6iB robot with an older R-j3iB controller. Some relevant information about the controller:

Core software version: Handling Tool 6.4
Socket Messaging (SKMG), includes MSG_CONNECT, MSG_DISCO and MSG_PING: Installed
PC Interface (PCIF): Installed
KAREL (KARL): Installed (not sure about the version, but I can check)
ASCII Upload (ASBN): Installed.
Is it possible to tweak the ROS driver and recompile it for R-j3iB? Maybe somebody has tried this already and can help.

Regards, 
Alex

gavanderhoorn, 2021-07-04:
As I replied on [your ROS Answers question](https://answers.ros.org/question/381676), we've had some people try to use `fanuc_driver` on R-J3iB and older.

The changes I made are still available here: https://github.com/ros-industrial/fanuc/compare/melodic-devel...gavanderhoorn:v611_karel_moveto.

That's for `V6.11` specifically, so it could be that `V6.40` doesn't need all of those, but this might form a good starting point.

Biggest differences with `HEAD`:

 - uses Karel `MOVE TO` statements for motion, instead of TP motions
 - doesn't use TP `RUN(..)`, but a small Karel program that calls `RUN_TASK(..)` (as the controller I created that branch for didn't have the multi-tasking option installed)
 - disables colours in log lines (no iPendant)
 - doesn't monitor all requirements for motion like current `HEAD` does

My recommendation would be to start with the state relay side: compile all the Karel programs and see whether the socket infrastructure works correctly in that. If you get no complaints (ie: Karel errors) try to use the [robot_state_visualize_m6ib.launch](https://github.com/ros-industrial/fanuc/blob/9e7662d2eebd894cf447712368a2c24657a97452/fanuc_m6ib_support/launch/robot_state_visualize_m6ib.launch) file to check whether what RViz shows corresponds to your real robot.

If that all works, there is a reasonable expectation the trajectory relay could work as well.

**NOTE**: make sure your controller only has a single motion group configured/active. Fanuc has been advising against using `MOVE TO` for years, and warning against using it on controllers with multiple motion groups. The branch I link to *does* use `MOVE TO`, so to avoid any surprises, only use it with controllers with a single motion group.

gavanderhoorn, 2021-07-04:
Reason to check state reporting first is also to make sure #105 isn't actually an issue on your controller.

RViz would not be able to correctly visualise your robot if there is a problem with the deg -> rad conversion.

gavanderhoorn, 2021-07-12:
@marabount: have you been able to test the branch I mentioned?

gavanderhoorn, 2021-07-12:
As this is not really an issue with the packages in this repository (yet), I'm going to convert this to a discussion.

We can open new issues if/when the need arises.

closed 2021-07-12 by gavanderhoorn