ros-industrial/fanuc #328: CRX-10iA/L Fanuc Driver Installation Issues

magnipopam, 2021-10-04:
We are trying to install the driver to our CRX-10iA/L which uses 30iB Mini Plus controller using the steps outlined in the following tutorial: http://wiki.ros.org/fanuc/Tutorials/hydro/Installation. We successfully installed the driver to our other CRX-10iA/L robot last year, but we are having issues installing it to our second CRX.

Here are the steps I've taken to install the driver in the order I've attempted these steps:

1. I uploaded the prebuilt binaries from the tutorial, but I received an error message about the version
2. I copied the binaries from the other controller which we've already successfully installed the driver on and uploaded those files to our robot, but I received the MEMO-159 error for 3 of the PC files (LIBIND_PKT, ROS_RELAY, and ROS_STATE). I followed the steps in the Troubleshooting page and deleted all the PC and VR/VA files generated by the upload with the exception of 2 VR files (LIBINDU_PKT and LIBSSOCK). When I try to delete those files, nothing happens, they still remain on the program list. Afterwards, when I tried reuploading the files, I received the same error message.
3. I cloned the binaries from source and compiled them using Roboguide. I  then uploaded those files to the controller. I received the same MEMO-159 error as in step 2. I attempted to delete the files and reupload, but I still receive that error. The two VR files also still cannot be deleted.

I am not really sure what the problem is. I am thinking it has to do with the 2 VR files that can't be deleted. The strange thing about those files is that I can only view them in the Program SELECT menu. When I look at the files in the Mem Device menu, I don't see them. I even tried viewing the files on my computer using an FTP and selecting "Show Hidden Files", but I still did not see those 2 files. I am thinking those 2 files are getting deleted, but for some their names are still in the program list.

gavanderhoorn, 2021-10-04:
I'm curious why you even get a `MEMO-159` if those files were never on the controller before your attempt to install them.

`.vr` files sticking around can be caused by *something* in the controller having references to them still open, despite the `.pc` files having already been removed.

In any case: I would suggest to connect to the FTP server, browsing to the `md:` device, sorting by extension and then looking for the two problematic `.vr` files.

If they are not there, do a power-cycle of the controller (ie: a cold start).

Then try again.

You never really "delete `.va` files" btw. Those are only for humans, and auto-generated by the controller. The order of operations would be to first delete the `.pc` file, which should take the `.vr` with it. If it didn't, manually delete the `.vr`. The `.va` should automatically disappear (it's not really a real file).

gavanderhoorn, 2021-10-06:
Did this work for you?

gavanderhoorn, 2021-10-10:
Closing due to inactivity.

closed 2021-10-10 by gavanderhoorn