ros-industrial/fanuc #329: changing registers

pzurk, 2022-02-12:


gavanderhoorn, 2022-02-12:
I've transferred your issue to this repository, as I believe this is a better place for it. `ros-industrial/industrial_core` hosts nothing Fanuc specific.

As to your question: no there is no "simply way to change registers" using the package(s) in this repository.

You could take a look at how `fanuc_driver` receives data from the ROS side and updates position registers with that data. You could then change that to use the same principle, but write to regular registers.

However, if you're using a recent controller, I'd probably suggest looking at something like Remote Motion Interface (or `R912`). It's an official Fanuc software option which lets you interact with the controller using JSON messages over a TCP connection.

Alternatively, you could see whether you could duplicate what [gavanderhoorn/dominh](https://github.com/gavanderhoorn/dominh) does. That's in Python, but the interfaces it uses can be used with C++ as well.

gavanderhoorn, 2022-02-12:
As this issue is largely off-topic, and it doesn't report any issues with the packages in this repository, I'm going to close it.

Feel free to keep commenting of course.

closed 2022-02-12 by gavanderhoorn